ms-iot/iot-adk-addonkit #253: Question for ARM Retail Images on Toolkit V4.4 - 16299.64 Windows Version

majidshaffi, 2018-07-10:
I am building a retail image and our custom app talks to the device through serial port and then later onboard itself to local wifi. Its only working when I add **IOT_TOOLKIT** feature in the RetailOEMInput file.
This is OK but It also allows me to connect using powershell and appears in IoT dashboard which I do not want.

**IOT_TOOLKIT opens which specific ports**? On Microsoft documentation it does not specify which ports it enables. Please refer to the below text:

"Includes developer tools such as: Kernel Debug components, FTP, Network Diagnostics, basic device portal, and XPerf. This also relaxes the firewall rules and enables various ports."

I don't want the device to show up on IoT dashboard and connect remotely using powershell. Please advice. My RetailOEMInput looks like this:

  <Features>
    <Microsoft>
      <Feature>IOT_EFIESP</Feature>
      <Feature>IOT_EFIESP_BCD_MBR</Feature>
      <Feature>IOT_DMAP_DRIVER</Feature>
      <Feature>IOT_GENERIC_POP</Feature>
      <Feature>IOT_CP210x_MAKERDRIVER</Feature>
      <Feature>IOT_FTSER2K_MAKERDRIVER</Feature>
      
      <Feature>IOT_UAP_OOBE</Feature>
      <Feature>IOT_APP_TOOLKIT</Feature>
      <Feature>**IOT_TOOLKIT**</Feature>
      <Feature>**IOT_POWERSHELL**</Feature>

endyb66, 2018-07-10:
@majidshaffi, i'm interested in creating a retail image, but i don't know what i need for it. I can build a test image successfully. Can you help me please and give me some advice on what i need and what i have to do different to test image?

majidshaffi, 2018-07-31:
@parameshbabu,
Can you please help me in this issue?

anphel31, 2018-08-09:
Hello,
The modified firewall ports are stored in plain text inside of the IOT_TOOLKIT package. Open the package (Microsoft-OneCore-IOTUAP-ToolKit-Package.cab) with a ZIP utility, then open the "...startup-settings...manifest" file inside the CAB.
You will see all of the firewall ports modified under the following registry key:
<registryKey keyName="HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Services\SharedAccess\Parameters\FirewallPolicy\FirewallRules">

anphel31, 2018-08-31:
Here is the registry key: HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Services\SharedAccess\Parameters\FirewallPolicy\FirewallRules

parameshbabu, 2018-11-16:
Closing this issue.

closed 2018-11-16 by parameshbabu