ms-iot/iot-adk-addonkit #254: Package creation fails

Nil0, 2018-07-11:
IOTADK_ROOT : C:\iot-adk-addonkit
ADK_VERSION : 17134
WDK_VERSION : 17134
COREKIT_VER : 17134.1

c:\bsp>build.cmd
signing c:\bsp\bcm2836pwm.sys
signing c:\bsp\bcm2836sdhc.sys
signing c:\bsp\bcmauxspi.sys
signing c:\bsp\bcmgpio.sys
signing c:\bsp\bcmi2c.sys
signing c:\bsp\bcmspi.sys
signing c:\bsp\pi_miniuart.sys
signing c:\bsp\rpiq.sys
signing c:\bsp\rpisdhc.sys
signing c:\bsp\rpiwav.sys
signing c:\bsp\SerPL011.sys
signing c:\bsp\vchiq.sys
signing c:\bsp\vchiq_arm_kern.dll
signing c:\bsp\vcos_win32_kern.dll
Building packages in c:\bsp\
 Processing bcm2836pwm.wm.xml
(PkgGenCommon) : error : Failed DoConvert
Error : Failed to create package. See C:\iot-adk-addonkit\Build\arm\pkgs\logs\bcm2836pwm.wm.log
 Processing bcm2836sdhc.wm.xml
(PkgGenCommon) : error : Failed DoConvert
Error : Failed to create package. See C:\iot-adk-addonkit\Build\arm\pkgs\logs\bcm2836sdhc.wm.log
 Processing bcmauxspi.wm.xml
(PkgGenCommon) : error : Failed DoConvert
Error : Failed to create package. See C:\iot-adk-addonkit\Build\arm\pkgs\logs\bcmauxspi.wm.log
 Processing bcmgpio.wm.xml
(PkgGenCommon) : error : Failed DoConvert
Error : Failed to create package. See C:\iot-adk-addonkit\Build\arm\pkgs\logs\bcmgpio.wm.log
 Processing bcmi2c.wm.xml
(PkgGenCommon) : error : Failed DoConvert
Error : Failed to create package. See C:\iot-adk-addonkit\Build\arm\pkgs\logs\bcmi2c.wm.log
 Processing bcmspi.wm.xml
(PkgGenCommon) : error : Failed DoConvert
Error : Failed to create package. See C:\iot-adk-addonkit\Build\arm\pkgs\logs\bcmspi.wm.log
 Processing piminiuart.wm.xml
(PkgGenCommon) : error : Failed DoConvert
Error : Failed to create package. See C:\iot-adk-addonkit\Build\arm\pkgs\logs\piminiuart.wm.log
 Processing rpiq.wm.xml
(PkgGenCommon) : error : Failed DoConvert
Error : Failed to create package. See C:\iot-adk-addonkit\Build\arm\pkgs\logs\rpiq.wm.log
 Processing rpisdhc.wm.xml
(PkgGenCommon) : error : Failed DoConvert
Error : Failed to create package. See C:\iot-adk-addonkit\Build\arm\pkgs\logs\rpisdhc.wm.log
 Processing rpiwav.wm.xml
(PkgGenCommon) : error : Failed DoConvert
Error : Failed to create package. See C:\iot-adk-addonkit\Build\arm\pkgs\logs\rpiwav.wm.log
 Processing SerPL011.wm.xml
(PkgGenCommon) : error : Failed DoConvert
Error : Failed to create package. See C:\iot-adk-addonkit\Build\arm\pkgs\logs\SerPL011.wm.log
 Processing SV.PlatExtensions.wm.xml
(PkgGenCommon) : error : Failed DoConvert
Error : Failed to create package. See C:\iot-adk-addonkit\Build\arm\pkgs\logs\SV.PlatExtensions.wm.log
 Processing vchiq.wm.xml
(PkgGenCommon) : error : Failed DoConvert
Error : Failed to create package. See C:\iot-adk-addonkit\Build\arm\pkgs\logs\vchiq.wm.log

IoTCoreShellv5.3 arm 10.0.0.0

anphel31, 2018-07-11:
Can you share output from one of the log files? (bcm2836pwm.wm.log)

Nil0, 2018-07-12:
Creating vchiq.wm.xml Package with version 10.0.0.0 for DigitalSign
Microsoft (C) PkgGen 10.0.10011.16384

--Stack Trace--
   at Microsoft.WindowsPhone.ImageUpdate.PkgGenCommon.InfFileConverter.DoConvert(String infFile, String[] references, String[] stagingSubdirs, String hiveRoot, String wimRoot, String targetPartition, CpuId cpu, String diffedHivesPath, String productName, String toolPaths)
   at Microsoft.CompPlat.PkgBldr.Plugins.WmToCsi.InfConverter.InstallDriver(List`1 driverCapabilityList, Boolean isUniveralBsp)
   at Microsoft.CompPlat.PkgBldr.Plugins.WmToCsi.Driver.ConvertEntries(XElement toCsi, Dictionary`2 plugins, Config environ, XElement driverWmElement)
   at Microsoft.CompPlat.PkgBldr.Base.PkgPlugin.ConvertEntries(XElement parent, Dictionary`2 plugins, Config enviorn, XElement component)
   at Microsoft.CompPlat.PkgBldr.Plugins.WmToCsi.Drivers.ConvertEntries(XElement toCsi, Dictionary`2 plugins, Config environ, XElement fromWm)
   at Microsoft.CompPlat.PkgBldr.Base.PkgPlugin.ConvertEntries(XElement parent, Dictionary`2 plugins, Config enviorn, XElement component)
   at Microsoft.CompPlat.PkgBldr.Plugins.WmToCsi.Identity.ConvertEntries(XElement ToCsi, Dictionary`2 plugins, Config enviorn, XElement FromWm)
   at Microsoft.CompPlat.PkgBldr.Program.BuildPackage(Config config, MacroResolver commandLineMacros)
   at Microsoft.CompPlat.PkgBldr.Program.Main(String[] args)
--End Stack Trace--
Package creation failed with error -1

anphel31, 2018-08-30:
This appears to be the same error as #229 
You could try these workaround steps:
Open: "Settings" -> "Time and Language" -> "Region and Language" -> "Administrative Language Settings" -> "Copy settings..." -> Select "Welcome screen and system accounts" -> click "Ok" to reboot system.

parameshbabu, 2018-10-08:
Windows 10 ADK for 1809 release has fix for the language issue. Please try with that version and this issue should be fixed. Closing this now.

closed 2018-10-08 by parameshbabu