ms-iot/iot-adk-addonkit #256: the system cannot find the file specified

KobeFleerackers, 2018-07-30:
When I do `inf2cab D:\_Kobe\driver\driver\oem71.inf` I first get:
```
Authoring Drivers.oem71.wm.xml
Processing Drivers.oem71.wm.xml
```
but then i get:
```
the system cannot find the file specified
ThreadId11364 ERROR: (0,0):
ERROR: 0x80070002
ERROR: PkgGen
ERROR: EXCEPTION: ExecuteCommands::Run: Failed to find makecat.exe. Make sure path environment variable is set right
ERROR: STACKTRACE:   bij Microsoft.Composition.ToolBox.ExecuteCommands.Run(String toolPath, String args, Boolean loadUserProfile, UInt32 retryCount, Logger logger)
ERROR:   bij Microsoft.Composition.Packaging.PkgManifest.CreateCatalog(String catPath, IEnumerable`1 filesToSign, String packageName, Keyform keyform, PackageType packageType)
ERROR:   bij Microsoft.Composition.Packaging.PkgManifest.GenerateAndSignCatalog(Boolean signCatalog)
ERROR:   bij Microsoft.Composition.Packaging.PkgManifest.SerializeMum(String outputFolder)
ERROR:   bij Microsoft.Composition.Packaging.PkgManifest.SaveManifest(String outputFolder)
ERROR:   bij Microsoft.Composition.Packaging.CbsPackage.SavePackage(String outputFolder, Boolean compressManifests)
ERROR:   bij Microsoft.Composition.Packaging.CbsPackage.SavePackage(String outputFolder)

(PkgToolBox) : error : ExecuteCommands::Run: Failed to find makecat.exe. Make sure path environment variable is set right
 Error : Failed to create package. See D:\_Kobe\driver\driver\Drivers.oem71.wm.log
```

In the log file I read the following:
```
Creating Drivers.oem70.wm.xml Package with version 10.0.1.0 for 
Microsoft (C) PkgGen 10.0.10011.16384

HKEY_LOCAL_MACHINE\SYSTEM\DriverDatabase\DriverPackages\oem70.inf_x86_af229e1a5336e4b5\Configurations\Minidriver32_61_Install.NT\Properties\{a45c254e-df1c-4efd-8020-67d146a850e0}\000E
D:(A;CI;KA;;;SY)
NT AUTHORITY\SYSTEM,FullControl

HKEY_LOCAL_MACHINE\SYSTEM\Setup\ResolveFilePaths
D:(A;OICI;KR;;;RC)(A;OICI;KA;;;SY)(A;OICI;KA;;;BA)
NT AUTHORITY\BEPERKT,ReadKey
NT AUTHORITY\SYSTEM,FullControl
INGEBOUWD\Administrators,FullControl

ThreadId6472 INFO 
Loading Package:C:\Users\Michel\AppData\Local\Temp\syslzvvj.dbd\update.mum\.Drivers.oem70~31bf3856ad364e35~x86~~10.0.1.0.mum
ThreadId6472 INFO Loading manifest file C:\Users\Michel\AppData\Local\Temp\syslzvvj.dbd\update.mum\.Drivers.oem70~31bf3856ad364e35~x86~~10.0.1.0.mum
ThreadId6472 INFO 
Done loading Package:C:\Users\Michel\AppData\Local\Temp\syslzvvj.dbd\update.mum\.Drivers.oem70~31bf3856ad364e35~x86~~10.0.1.0.mum
ThreadId6472 INFO Adding file of type CsiManifest with detination path  and source path C:\Users\Michel\AppData\Local\Temp\cpufor3p.ube\manifests\.Drivers.oem70.man
ThreadId6472 INFO Converting C:\Users\Michel\AppData\Local\Temp\cpufor3p.ube\manifests\.Drivers.oem70.man to CMI manifest
ThreadId6472 INFO Saving CMI Manifest File to : C:\Users\Michel\AppData\Local\Temp\ts1zo1md.thw
ThreadId6472 INFO Adding file of type Manifest with detination path  and source path C:\Users\Michel\AppData\Local\Temp\ts1zo1md.thw\x86_.drivers.oem70_31bf3856ad364e35_10.0.1.0_none_6e00b92b9c937322.manifest
ThreadId6472 INFO Loading manifest file C:\Users\Michel\AppData\Local\Temp\q24qwfnq.hul\x86_.drivers.oem70-deployment_31bf3856ad364e35_10.0.1.0_none_d8eada46b31a0442.manifest
Found Driver = C:\Users\Michel\AppData\Local\Temp\cpufor3p.ube\drivers\oem70.inf
ThreadId6472 INFO Adding file of type CsiManifest with detination path  and source path C:\Users\Michel\AppData\Local\Temp\cpufor3p.ube\manifests\dual_oem70.inf.man
ThreadId6472 INFO Converting C:\Users\Michel\AppData\Local\Temp\cpufor3p.ube\manifests\dual_oem70.inf.man to CMI manifest
ThreadId6472 INFO Copying 'D:\_Kobe\driver\driver\oem70.inf' to 'C:\Users\Michel\AppData\Local\Temp\4yykdphz.aof\x86_dual_oem70.inf_31bf3856ad364e35_10.0.1.0_none_97edf2b190a48552\oem70.inf'.
ThreadId6472 INFO Saving CMI Manifest File to : C:\Users\Michel\AppData\Local\Temp\4yykdphz.aof
ThreadId6472 INFO Adding file of type Manifest with detination path  and source path C:\Users\Michel\AppData\Local\Temp\4yykdphz.aof\x86_dual_oem70.inf_31bf3856ad364e35_10.0.1.0_none_97edf2b190a48552.manifest
ThreadId6472 INFO 
Saving CAB D:\Build\x86\pkgs\.Drivers.oem70.cab for package .Drivers.oem70
ThreadId6472 INFO 
Saving package .Drivers.oem70 to C:\Users\Michel\AppData\Local\Temp\t0zrnkcl.wnw
ThreadId6472 INFO 
Validating Package '.Drivers.oem70'
ThreadId6472 INFO Serialized C:\Users\Michel\AppData\Local\Temp\t0zrnkcl.wnw\x86_-drivers-oem70-deployment-deployment_31bf3856ad364e35_10.0.1.0_none_485b13d3c5ee921e.manifest to disk
ThreadId6472 INFO Running makecat.exe /v "C:\Users\Michel\AppData\Local\Temp\jq3yfkv5.3y1\.Drivers.oem70~31bf3856ad364e35~x86~~10.0.1.0"
--Stack Trace--
   bij Microsoft.Composition.ToolBox.ExecuteCommands.Run(String toolPath, String args, Boolean loadUserProfile, UInt32 retryCount, Logger logger)
   bij Microsoft.Composition.Packaging.PkgManifest.CreateCatalog(String catPath, IEnumerable`1 filesToSign, String packageName, Keyform keyform, PackageType packageType)
   bij Microsoft.Composition.Packaging.PkgManifest.GenerateAndSignCatalog(Boolean signCatalog)
   bij Microsoft.Composition.Packaging.PkgManifest.SerializeMum(String outputFolder)
   bij Microsoft.Composition.Packaging.PkgManifest.SaveManifest(String outputFolder)
   bij Microsoft.Composition.Packaging.CbsPackage.SavePackage(String outputFolder, Boolean compressManifests)
   bij Microsoft.Composition.Packaging.CbsPackage.SavePackage(String outputFolder)
   bij Microsoft.Composition.Packaging.CbsPackage.SaveCab(String cabPath, CompressionType compressionType)
   bij Microsoft.CompPlat.PkgBldr.CabGen.Assembly.ConvertEntries(XElement parent, Dictionary`2 plugins, Config enviorn, XElement component)
   bij Microsoft.CompPlat.PkgBldr.Program.BuildPackage(Config config, MacroResolver commandLineMacros)
   bij Microsoft.CompPlat.PkgBldr.Program.Main(String[] args)
--End Stack Trace--
Package creation failed with error -1
```

I have no clue what the problem is here.

Kobe118, 2018-07-31:
solved: [https://stackoverflow.com/questions/51594544/error-when-making-a-driver-for-windows-iot-core-using-wdk?noredirect=1#comment90183162_51606914](https://stackoverflow.com/questions/51594544/error-when-making-a-driver-for-windows-iot-core-using-wdk?noredirect=1#comment90183162_51606914)

anphel31, 2018-07-31:
Sounds like you resolved this error by installing the WDK as well as the Windows 10 IoT Core Package kits.
Can you confirm?

closed 2018-08-08 by parameshbabu