ms-iot/iot-adk-addonkit #257: Test recovery partition

TerrenceHenry, 2018-07-31:
How am I able to test the recovery partition.
I have built Flash_Recovery.ffu, and it boots fine.
From the startup sequence, I can see that's is different to Flash.ffu

I need to be able to verify that recovery will work, so I would like to be able to 
1. Force the recovery to happen
2. Delete some file that will cause recovery to happen.

Please advise.

Thanks in advance

anphel31, 2018-08-03:
Try wrapping the entire command in quotes, like this:
"bcdedit /set {bootmgr} bootsequence {a5935ff2-32ba-4617-bf36-5ac314b3f9bf}"

TerrenceHenry, 2018-08-03:
When I wrap it in quotes, I don't get the "..parameter was already specified." response, but the reboot looks exactly like any other, so I still don't see any indication of  a recovery happening.

anphel31, 2018-08-30:
Did you log into the Administrator account with Powershell?
Can you run "Bcdedit /enum all" after running the /set command, and confirm that the bootsequence entry is being changed to  {a5935ff2-32ba-4617-bf36-5ac314b3f9bf} ?

parameshbabu, 2018-11-16:
Closing. If the issue is still open, please provide more details and also try asking at the https://social.msdn.microsoft.com/forums/en-US/home?forum=WindowsIoT

closed 2018-11-16 by parameshbabu