ms-iot/iot-adk-addonkit #258: Flash_Recovery.ffu causes screen orentation problems

TerrenceHenry, 2018-08-02:
I use Win 10 IOT on a Raspberry Pi 3b.
I use a Pi touch screen, which is configured as Landscape Flipped, via the Windows Device Portal.
Win 10 IOT version is v.10.0.17134.1

When using Flash_Recovery.ffu (recovery build of Win 10 IOT), it does the following:
1. On boot, the Windows icon and progress circle are upside down. 
Looks like the flag controlling the screen orientation is not read by the boot-loader.
When the application is launched, the orientation is correct.

2. When accessing the device via the windows device portal, capturing a screenshot shows the screen upside down.
This is incorrect.

anphel31, 2018-08-30:
Can you try running the flash.ffu image (not the recovery version) to see if this is a recovery related issue or some more generic issue?

parameshbabu, 2018-11-16:
Since the image has the orientation inverted, applying the recovery image also gets the same behavior. Instead of setting the configuration via Windows Device Portal, this configuration should be set in the image itself to avoid this issue.

closed 2019-04-17 by saraclay