ms-iot/iot-adk-addonkit #260: ERROR: Please run imageapp.exe in an elevated console.

jermeyjungbeker, 2018-08-17:
```
IOTADK_ROOT : C:\Users\admin\OneDrive\Desktop\iot-adk-addonkit-master
ADK_VERSION : 17134
WDK_VERSION : 17134
COREKIT_VER : 10586.0
OEM_NAME    : MMC

Set Environment for Architecture
Choose 1 for ARM, 2 for x86 and 3 for x64 : 1

Configuring for arm architecture
KitsRoot       : [C:\Program Files (x86)\Windows Kits\10\]
WDKContentRoot : []
Retail Signing disabled
BSP_ARCH    : arm
BSP_VERSION : 10.0.0.0
BSPPKG_DIR  : C:\Users\admin\OneDrive\Desktop\iot-adk-addonkit-master\Build\arm\pkgs
MSPKG_DIR   : C:\Program Files (x86)\Windows Kits\10\MSPackages\Retail\arm\fre

IoTCoreShellv5.3 arm 10.0.0.0
C:\Users\admin\OneDrive\Desktop\iot-adk-addonkit-master\Tools>buildimage ProductA test
Creating ProductA test Image, see C:\Users\admin\OneDrive\Desktop\iot-adk-addonkit-master\Build\arm\ProductA_test.log for progress
ThreadId17528 ERROR: Please run imageapp.exe in an elevated console.

IoTCoreShellv5.3 arm 10.0.0.0
C:\Users\admin\OneDrive\Desktop\iot-adk-addonkit-master\Tools>
```

I'm running it as administrator and i am working on Windows 10, version 1803 . I'm trying to follow https://docs.microsoft.com/en-us/windows-hardware/manufacture/iot/create-a-basic-image. I have tried SampleA as well and i get the exact same error and when i look at the log the only line is the error specified above.

parameshbabu, 2018-08-29:
COREKIT_VER : 10586.0 - The core kit version is very old. Please uninstall and install the latest IoT Core Kit version. 
Here is the fix for the elevated console issue, if you are using non-EN-US language.

We've been able to reproduce this error after configuring a machine with non-EN-us localized version of Windows 10. The issue should be fixed in the next kits release, but in the meantime, here is one way to work around the issue. Open the Computer Management console (compmgmt.msc), select "Local Users and Groups" -> "Groups". Find the Administrators group (which would be localized to another language); rename this group to the English spelling: "Administrators", and then rerun the image creation steps. After image creation you can return the name back to the original spelling

jermeyjungbeker, 2018-09-03:
I got windows 10 home so i can't do what you suggest. Also i think COREVIT_VER 1586 is what it defaults to even if i put the latest version in the map: C:\Program Files (x86)\Windows Kits\10\MSPackages\Retail or C:\Program Files (x86)\Windows Kits\10\MSPackages\Retail\arm\fre

parameshbabu, 2018-10-04:
Please try with again with 1809 ADK tools.

jermeyjungbeker, 2018-10-05:
Got it working a few weeks ago by using my computer at home just forgot to close this issue.

closed 2018-10-05 by jermeyjungbeker