ms-iot/iot-adk-addonkit #261: Custom.cmd, Provisioning.Auto and Recovery.WinPE not getting built

munsingh, 2018-08-29:
From the master branch buildpkg all is skipping the above Packages, because of which buildfm oem is failing:

Contents of buildfm_oem.log file:
--------------------------------------------------------------------------------
info: Trying to load file 'C:\iot-adk-addonkit-master\Build\amd64\InputFMs\OEMFMFileList.xml' as a FM file list ...
fatal error : Error: Missing package: C:\iot-adk-addonkit-master\Build\amd64\pkgs\AMD.Custom.Cmd.cab
fatal error : Error: Missing package: C:\iot-adk-addonkit-master\Build\amd64\pkgs\AMD.Custom.Cmd.cab
fatal error : Error: Missing package: C:\iot-adk-addonkit-master\Build\amd64\pkgs\AMD.Provisioning.Auto.cab
fatal error : Error: Missing package: C:\iot-adk-addonkit-master\Build\amd64\pkgs\AMD.Provisioning.Auto.cab
fatal error : Error: Missing package: C:\iot-adk-addonkit-master\Build\amd64\pkgs\AMD.Recovery.WinPE.cab
--------------------------------------------------------------------------------

I also see a new folder inside Common called ProdPackages, which was not there in the 4.4 branch.

Please fix it as because of this buildfm oem is not working.

HalHunt, 2018-08-29:
I'm seeing the same errors.  For me, I originally set my OEM_NAME to "Hyprsoft Corporation" and got other errors during build.  I thought it may be because I had a space in the OEM_NAME.  However the packages mentioned above **were** being built properly.  I decided to delete my entire "iot-adk-addonkit" folder and started over from scratch with a new OEM_NAME of just "Hyprsoft" hoping to rectify my original errors.  Now I am seeing the errors mentioned above.  If I try to build an individual package I get:

D:\iot\iot-adk-addonkit>buildpkg Custom.Cmd
 Processing Custom.Cmd.wm.xml
(ToolsCommon) : error : Illegal characters in path.

and here is the log:

Creating Custom.Cmd.wm.xml Package with version 10.0.0.0 for DigitalSign
Microsoft (C) PkgGen 10.0.10011.16384
--Stack Trace--
   at Microsoft.WindowsPhone.ImageUpdate.Tools.Common.CmdArgsParser.ParseArgs[T](List`1 args, Object[] configuration)
   at Microsoft.CompPlat.PkgBldr.Program.Main(String[] args)
--End Stack Trace--
Package creation failed with error -1

munsingh, 2018-08-30:
Thanks Paramesh. I figured this out, went ahead and called createimage and saw that it indeed called buildfm and everything worked. 

Thanks for the response.

-Regards,
Manish

closed 2018-08-30 by munsingh