ms-iot/iot-adk-addonkit #268: File Permissions/TrustedInstaller Question

HalHunt, 2018-09-28:
When we include a custom package in our custom RPI3 FFU that contains files, the files are flashed onto the device with read and execute (RX) file permissions only.   This includes all accounts on the device (ALL APPLICATION PACKAGES, ALL RESTRICTED APPLICATION PACKAGES, the Adminstrators group, Users, and SYSTEM) **EXCEPT** TrustedInstaller which has full control (F) access permissions.  We've tried to run takeown.exe and icacls.exe in the oemcustomization.cmd without success.  **How can we grant SYSTEM full control file permissions without requiring human intervention?**  We have also tried flashing the files into different folders on the device like c:\xxx and c:\data\xxx without success.  **Is there a folder on the device more suitable for this behavior?**  The testing process to try different solutions is very cumbersome so we are posting here for help or any guidance.  Thank you in advance.

parameshbabu, 2018-11-16:
This is by design ensuring that the files you deliver are intact and can be restored when required. Can you explain what you are trying to do here and we can try to address that. Recommend to post that query at https://social.msdn.microsoft.com/forums/en-US/home?forum=WindowsIoT

HalHunt, 2018-12-18:
@majidshaffi,
Unfortunately no, we were unable to get around the file permission problem.  In our use case we had a Windows service that we used to remotely update .NET Core apps.  Instead of including the .NET Core binaries in the custom FFU we just let the service download and install the app the first time the device is turned on.  Here is the repo if it helps: [IoTCoreAppUpdates](https://github.com/hyprsoftcorp/IoTCoreAppUpdates).

saraclay, 2019-04-17:
If this is still an issue, I encourage you to post your question in our forums, where we have dedicated staff there to help you out. Thank you.

closed 2019-04-17 by saraclay