ms-iot/iot-adk-addonkit #272: IoTOnboardingTask missing after creating image

endyb66, 2018-10-02:
When building a new image for the Dragonboard 410c, the IoTOnboardingTask is missing after flashing the image to the device. Instead the old AllJoyn app is there. I followed the guide on MSDN to build an image, did i miss something? Also the new IoTCoreDefaultApp is no there, still the older version only. But this will be updated to the newer one.

parameshbabu, 2018-10-02:
The sample apps binaries have not been updated yet. While we update the binaries, you can replace them by rebuilding the corresponding projects from ms-iot/samples github.

endyb66, 2018-10-02:
The IoTCoreDefaultApp from MS-iot/samples is 11 months old, so it seems this also is the older one. Also the IoTOnboardingTask project seems to be an older one there.

parameshbabu, 2018-10-04:
Are you referring to the SoftAP functionality, then you should include IOT_ONBOARDING_APP feature id. The sample doesn't contain this and i'll have that updated.

endyb66, 2018-10-05:
Ok thank you.

closed 2018-10-09 by parameshbabu