ms-iot/iot-adk-addonkit #284: Powershell SignTool Error: No certificates were found that met all the given criteria.

scout208, 2018-10-11:
Trying to run `Add-IoTSignature C:\MyWorkspace *.sys,*.dll,*.exe` or `buildpkg All` in IoTCorePShell both fail with 

    SignTool Error: No certificates were found that met all the given criteria.

These commands don't fail in the regular IoTCoreShell.

parameshbabu, 2018-10-11:
Are you using latest ADK (1809?)

scout208, 2018-10-11:
No, I wasn't able to use the latest ADK because my IoT Core version is not latest, it's 10.0.17134.1.

parameshbabu, 2018-10-11:
i recommend to use the 17134 branch.

parameshbabu, 2018-10-11:
#285 should help to use the latest scripts with RS4 ADK. Let me know if this fix solves your issue.

scout208, 2018-10-12:
Hi I tried using the latest version, but am still having the same issue. I wonder if my certificate isn't setup right?

parameshbabu, 2018-10-12:
you should have Windows OEM Test Cert 2017 (TEST ONLY) certs installed on your machine to run this. this is part of the Windows ADK, and you can install this by running "installoemcerts" in the powershell command line.

scout208, 2018-10-12:
Okay I just confirmed that when in test mode the command doesn't display any errors. It's only when I run `Set-IoTRetailSign On` and then try to run the command that I see the errors. So, I think maybe there's a problem with configuring my retail certificates. 

This is the segment of my xml:

    
    
    <EVSignToolParam>/s my /i "Entrust Extended Validation Code Signing CA - EVCS1" /n "Monarc, Inc" /fd SHA256</EVSignToolParam>
    
    
    <RetailSignToolParam>/s my /i "Entrust Extended Validation Code Signing CA - EVSC1" /n "Monarc, Inc" /ac "C:\Users\ConnorEarly\Downloads\Certificate.cer" /fd SHA256</RetailSignToolParam>

Also, to be clear, I'm trying to follow the [Device Update Center User Guide](https://docs.microsoft.com/en-us/windows-hardware/service/iot/using-device-update-center)

parameshbabu, 2018-10-12:
Is the retail cert installed in the machine in the user store or local machine? If it’s installed in the local machine you may want to try ‘ /sm’ instead of ‘/s my’

Regards,
Paramesh

scout208, 2018-10-12:
The certificate is installed in the user store

scout208, 2018-10-17:
I tried specifying the thumbprint and it's still not working for me. I've verified that the certificate is listed in the certificate store in Microsoft Management Console. I'm not sure what the problem is.

scout208, 2018-10-18:
I didn't realize that I had to have the USB Token from my provider plugged into my computer in order to use the code-signing certificate. That was my issue. After plugging in the token and running the command it prompted me for the token password and then I was able to use it.

closed 2018-10-18 by scout208