ms-iot/iot-adk-addonkit #287: Inconsistent use of sign.cmd from addon-kit

ScapinFranco, 2018-11-06:
Recently **sign.cmd** was added to iot-adk-addonkit.

From what I can see, it does not always use this version, it seems it also uses the sign.cmd from "C:\Program Files (x86)\Windows Kits\10\tools\bin\i386\sign.cmd".

I figured this because I changed sign.cmd to use another timestamp server, as the default server "http://timestamp.verisign.com/scripts/timestamp.dll" is not working reliable at all today (from my place).

parameshbabu, 2018-11-13:
@ScapinFranco  can you point to the failure cases to help me investigate this further? The sign.cmd is in the search path and put ahead of the installed version, so it should never pick the other one unless its referred to with absolute path.

ScapinFranco, 2018-11-14:
@parameshbabu Scrap that. It looks like I was looking at logs from previous build attempts leading me to think the old timestamp server is still used,
Anyways, it would be nice, if after a failed sign, - because timestamp server did not respond or response was corrupted, or whatever reason the internet had a hickup,- it would retry at least 2 times before giving up.

closed 2018-11-14 by parameshbabu