ms-iot/iot-adk-addonkit #289: Package creation fails on New-IoTFFUImage

axeldietz, 2018-11-13:
I followed the Lab 1a: Create a basic image sample and succeeded to reach the final step in order to create the FFU..

I had to use the workaround to revent the error as described in #229 Open: "Settings" -> "Time and Language" -> "Region and Language" -> "Administrative Language Settings" -> "Copy settings..." -> Select "Welcome screen and system accounts" -> click "Ok" to reboot system.

However the creation of the FFU image failed with the below error.

PS C:\MyWorkspace>New-IoTFFUImage ProductA Test
ADK_VERSION : 10.0.17134.1
IOTCORE_VER : 10.0.17763.107
BSP_VERSION : 10.0.0.0
HostOS Info : Microsoft Windows 10 Enterprise - 10.0.17134 - nl-NL en-US
Validating product feature ids
Reading feature ids in C:\MyWorkspace\Source-arm\BSP\Rpi2\Packages\RPi2FM.xml
Reading feature ids in C:\MyWorkspace\Common\Packages\OEMCommonFM.xml
Reading feature ids in C:\MyWorkspace\Source-arm\Packages\OEMFM.xml
Checking Microsoft features in OEMInput file..
Warning: IOT_APPLICATIONS is not defined
Warning: IOT_GENERIC_POP is not defined
Checking OEM features in OEMInput file..
Building product specific packages
Processing Registry.Version.wm.xml
Processing Custom.Cmd.wm.xml
Processing Provisioning.Auto.wm.xml
Building FM files..
Exporting OEM FM files..
Processing OEMFMList..
Exporting RPi2 BSP FM files
Processing RPi2FMFileList.xml
Creating Image..
See C:\MyWorkspace\Build\arm\ProductA_Test.log for progress
This will take a while...
ThreadId10288 ERROR: System.ArgumentNullException: Value cannot be null.
Parameter name: path2
   at System.IO.Path.Combine(String path1, String path2)
   at Microsoft.WindowsPhone.FeatureAPI.PkgFile.get_PackagePath()
   at Microsoft.WindowsPhone.FeatureAPI.FeatureManifest.FMPkgInfo..ctor(PkgFile pkg, String groupValue)
   at Microsoft.WindowsPhone.FeatureAPI.FeatureManifest.GetAllPackagesByGroups(List`1 supportedUILanguages, List`1 supportedLocales, List`1 supportedResolutions, List`1 supportedWowCpuTypes, String buildType, String cpuType, String msPackageRoot)
   at Microsoft.WindowsPhone.FeatureAPI.FeatureManifest.GetFilteredPackagesByGroups()
   at Microsoft.WindowsPhone.Imaging.Imaging.ProcessFMEntries(String fmFileXML, Boolean coreFM)
   at Microsoft.WindowsPhone.Imaging.Imaging.ProcessFMs()
   at Microsoft.WindowsPhone.Imaging.Imaging.GenerateInputFile(List`1 packages)
   at Microsoft.WindowsPhone.Imaging.Imaging.SelectPackagesToImage()
   at Microsoft.WindowsPhone.Imaging.Imaging.CreateFullImage(Boolean randomizeGptIds, Boolean recovery, Guid imagingSessionId)
   at Microsoft.WindowsPhone.Imaging.Imaging.ProcessImage(Boolean randomizeGptIds, Boolean recovery)
ThreadId10288 ERROR: An unhandled exception was thrown: System.ArgumentNullException: Value cannot be null.
Parameter name: path2
   at System.IO.Path.Combine(String path1, String path2)
   at Microsoft.WindowsPhone.FeatureAPI.PkgFile.get_PackagePath()
   at Microsoft.WindowsPhone.FeatureAPI.FeatureManifest.FMPkgInfo..ctor(PkgFile pkg, String groupValue)
   at Microsoft.WindowsPhone.FeatureAPI.FeatureManifest.GetAllPackagesByGroups(List`1 supportedUILanguages, List`1 supportedLocales, List`1 supportedResolutions, List`1 supportedWowCpuTypes, String buildType, String cpuType, String msPackageRoot)
   at Microsoft.WindowsPhone.FeatureAPI.FeatureManifest.GetFilteredPackagesByGroups()
   at Microsoft.WindowsPhone.Imaging.Imaging.ProcessFMEntries(String fmFileXML, Boolean coreFM)
   at Microsoft.WindowsPhone.Imaging.Imaging.ProcessFMs()
   at Microsoft.WindowsPhone.Imaging.Imaging.GenerateInputFile(List`1 packages)
   at Microsoft.WindowsPhone.Imaging.Imaging.SelectPackagesToImage()
   at Microsoft.WindowsPhone.Imaging.Imaging.CreateFullImage(Boolean randomizeGptIds, Boolean recovery, Guid imagingSessionId)
   at Microsoft.WindowsPhone.Imaging.Imaging.ProcessImage(Boolean randomizeGptIds, Boolean recovery)
Error: Build failed
False
IoTCorePShell arm 10.0.0.0 Test

Flash.ImageApp:

{11/13/2018 11:33:48 PM}  Imaging: Reading the OEM Input XML file...
{11/13/2018 11:33:48 PM}  FeatureAPI: Successfully validated the OEM Input XML: C:\MyWorkspace\Source-arm\Products\ProductA\TestOemInput.xml
{11/13/2018 11:33:48 PM}  Imaging: Reading the FM XML file 'C:\MyWorkspace\Build\arm\MergedFMs\RPi2FM.xml'...
{11/13/2018 11:33:48 PM}  FeatureAPI: Successfully validated the Feature Manifest XML: C:\MyWorkspace\Build\arm\MergedFMs\RPi2FM.xml
{11/13/2018 11:33:49 PM}  Imaging: Reading the FM XML file 'C:\MyWorkspace\Build\arm\MergedFMs\OEMCommonFM.xml'...
{11/13/2018 11:33:49 PM}  FeatureAPI: Successfully validated the Feature Manifest XML: C:\MyWorkspace\Build\arm\MergedFMs\OEMCommonFM.xml
{11/13/2018 11:33:49 PM}  Imaging: Reading the FM XML file 'C:\MyWorkspace\Build\arm\MergedFMs\OEMFM.xml'...
{11/13/2018 11:33:49 PM}  FeatureAPI: Successfully validated the Feature Manifest XML: C:\MyWorkspace\Build\arm\MergedFMs\OEMFM.xml
{11/13/2018 11:33:49 PM}  Imaging: Reading the FM XML file 'C:\Program Files (x86)\Windows Kits\10\FMFiles\arm\IoTUAPNonProductionPartnerShareFM.xml'...
{11/13/2018 11:33:49 PM}  FeatureAPI: Successfully validated the Feature Manifest XML: C:\Program Files (x86)\Windows Kits\10\FMFiles\arm\IoTUAPNonProductionPartnerShareFM.xml
{11/13/2018 11:33:49 PM} Error: System.ArgumentNullException: Value cannot be null.
Parameter name: path2
   at System.IO.Path.Combine(String path1, String path2)
   at Microsoft.WindowsPhone.FeatureAPI.PkgFile.get_PackagePath()
   at Microsoft.WindowsPhone.FeatureAPI.FeatureManifest.FMPkgInfo..ctor(PkgFile pkg, String groupValue)
   at Microsoft.WindowsPhone.FeatureAPI.FeatureManifest.GetAllPackagesByGroups(List`1 supportedUILanguages, List`1 supportedLocales, List`1 supportedResolutions, List`1 supportedWowCpuTypes, String buildType, String cpuType, String msPackageRoot)
   at Microsoft.WindowsPhone.FeatureAPI.FeatureManifest.GetFilteredPackagesByGroups()
   at Microsoft.WindowsPhone.Imaging.Imaging.ProcessFMEntries(String fmFileXML, Boolean coreFM)
   at Microsoft.WindowsPhone.Imaging.Imaging.ProcessFMs()
   at Microsoft.WindowsPhone.Imaging.Imaging.GenerateInputFile(List`1 packages)
   at Microsoft.WindowsPhone.Imaging.Imaging.SelectPackagesToImage()
   at Microsoft.WindowsPhone.Imaging.Imaging.CreateFullImage(Boolean randomizeGptIds, Boolean recovery, Guid imagingSessionId)
   at Microsoft.WindowsPhone.Imaging.Imaging.ProcessImage(Boolean randomizeGptIds, Boolean recovery)
{11/13/2018 11:33:49 PM} Error: An unhandled exception was thrown: System.ArgumentNullException: Value cannot be null.
Parameter name: path2
   at System.IO.Path.Combine(String path1, String path2)
   at Microsoft.WindowsPhone.FeatureAPI.PkgFile.get_PackagePath()
   at Microsoft.WindowsPhone.FeatureAPI.FeatureManifest.FMPkgInfo..ctor(PkgFile pkg, String groupValue)
   at Microsoft.WindowsPhone.FeatureAPI.FeatureManifest.GetAllPackagesByGroups(List`1 supportedUILanguages, List`1 supportedLocales, List`1 supportedResolutions, List`1 supportedWowCpuTypes, String buildType, String cpuType, String msPackageRoot)
   at Microsoft.WindowsPhone.FeatureAPI.FeatureManifest.GetFilteredPackagesByGroups()
   at Microsoft.WindowsPhone.Imaging.Imaging.ProcessFMEntries(String fmFileXML, Boolean coreFM)
   at Microsoft.WindowsPhone.Imaging.Imaging.ProcessFMs()
   at Microsoft.WindowsPhone.Imaging.Imaging.GenerateInputFile(List`1 packages)
   at Microsoft.WindowsPhone.Imaging.Imaging.SelectPackagesToImage()
   at Microsoft.WindowsPhone.Imaging.Imaging.CreateFullImage(Boolean randomizeGptIds, Boolean recovery, Guid imagingSessionId)
   at Microsoft.WindowsPhone.Imaging.Imaging.ProcessImage(Boolean randomizeGptIds, Boolean recovery)
{11/13/2018 11:33:49 PM}  Imaging: See Flash.cbs.log for details.
{11/13/2018 11:33:49 PM}  Imaging: Performance Results:
{11/13/2018 11:33:49 PM}  	Total Run Time:	00:00:01.3282698
{11/13/2018 11:33:49 PM}  	Storage Stack Time:	00:00:00.0000003
{11/13/2018 11:33:49 PM}  	Total run time:	1328.2698ms
{11/13/2018 11:33:49 PM}  	Total staging time:	0ms
{11/13/2018 11:33:49 PM}  	Total commit time:	0ms
{11/13/2018 11:33:49 PM}  	CompDB Total Time (ms):	791.9364ms
{11/13/2018 11:33:49 PM}  	Image format:	FFU
{11/13/2018 11:33:49 PM}  	Stage file time:	0
{11/13/2018 11:33:49 PM}  	UpdateOS time:	0
{11/13/2018 11:33:49 PM}  	CompDB Total Time:	00:00:00.7919364

Any ideas?

parameshbabu, 2018-11-14:
I see that you are using incompatible versions of the tools/packages.
ADK_VERSION : 10.0.17134.1
IOTCORE_VER : 10.0.17763.107
You should be installing 17763 version of ADK (Latest) and with that you dont need your workaround anymore.

axeldietz, 2018-11-14:
You are right two times! Thanks for your quick and appropriate reply.

I was wrongly assuming that the Windows ADK for Windows 10, version 1809 was referring to the host OS.

closed 2018-11-14 by axeldietz