ms-iot/iot-adk-addonkit #291: When run  Build packages getting this error

rtarta, 2018-11-16:
[image]

As like this  url says , i did as  what steps says. Now  when i run buildpkg All getting error as image.
https://docs.microsoft.com/tr-tr/windows-hardware/manufacture/iot/create-a-basic-image

rtarta, 2018-11-16:
Also this is log

Microsoft (C) PkgGen 10.0.10011.16384

--Stack Trace--
   at Microsoft.CompPlat.PkgBldr.Base.PkgGenResources.GetCommonSchemaStream(String schemaFileName)
   at Microsoft.CompPlat.PkgBldr.Base.PkgGenResources.GetSchemaStream(String schemaFileName, IPkgPlugin plugin)
   at Microsoft.CompPlat.PkgBldr.Base.SchemaSet.MergeSchema(List`1 schemaFileList, List`1 pluginList)
   at Microsoft.CompPlat.PkgBldr.Base.SchemaSet.LoadSchemasFromPlugins(IEnumerable`1 plugins)
   at Microsoft.CompPlat.PkgBldr.Base.PkgBldrLoader.LoadWmSchemas()
   at Microsoft.CompPlat.PkgBldr.Base.PkgBldrLoader.LoadPlugins()
   at Microsoft.CompPlat.PkgBldr.Base.PkgBldrLoader..ctor(PluginType pluginType, PkgBldrCmd pkgBldrArgs, IDeploymentLogger logger)
   at Microsoft.CompPlat.PkgBldr.Program.BuildPackage(Config config, MacroResolver commandLineMacros)
   at Microsoft.CompPlat.PkgBldr.Program.Main(String[] args)
--End Stack Trace--

parameshbabu, 2018-11-16:
Can you try running ‘installoemcerts’ cmd?
Maybe the certs are not installed properly.

Regards,
Paramesh

rtarta, 2018-11-16:
@parameshbabu thanks for your fast reply,  when i run the **installoemcerts** it says  certificates  already in store
after this command i run the **buildpkg All** bu same result 

[image]

parameshbabu, 2018-11-17:
Looks like the binaries are mixed up. Do you have older version of WDK still installed?

Best way forward will be to uninstall both ADK and WDK and reinstall the latest version of ADK.

Regards,
Paramesh

[image]<https://na01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fuser-images.githubusercontent.com%2F16166798%2F48646275-7098f100-e9f9-11e8-9254-3b16eb61c2b8.png&data=02%7C01%7Cpabab%40microsoft.com%7C01761c9158014e8d6fc008d64c04370f%7C72f988bf86f141af91ab2d7cd011db47%7C1%7C0%7C636779978319567315&sdata=a17ZoHqBgc%2BSpn1A%2FMQfU0znOamqkX%2F3Vg3CoU0vKM0%3D&reserved=0>

rtarta, 2018-11-17:
@parameshbabu  below my adk and iot core version numbers ?
is it true or not 
thanks.

[image]

parameshbabu, 2018-11-19:
Yes the versions displayed are correct. can you confirm that you dont have older WDK installed in your machine? This issue seems to be due to incorrect/incomplete install of the ADK.

parameshbabu, 2018-11-26:
@rtarta can you rerun the tool with the latest adk-addonkit tools ? Also can you provide your OS information?

rtarta, 2018-11-27:
@parameshbabu  hi i download the latest and rerun the commands as below my operation system is Windows Server 2012 R2. 

Loading IoTCoreImaging module..
Opening workspace : C:\iot-adk-addonkit-master\Workspace\IoTWorkspace.xml
Corekit found OK
Test certs installed
ADK_VERSION : 10.0.17763.1
IOTCORE_VER : 10.0.17763.107
BSP_VERSION : 10.0.0.0
ADDONKITVER : 6.0.181121.1730
HostOS Info : Microsoft Windows Server 2012 R2 Standard - 6.3.9600 - en-US
IOTWKSPACE  : C:\iot-adk-addonkit-master\Workspace
OEM_NAME    : Contoso
BSP_ARCH    : arm
BSPPKG_DIR  : C:\iot-adk-addonkit-master\Workspace\Build\arm\pkgs
MSPKG_DIR   : C:\Program Files (x86)\Windows Kits\10\MSPackages\Retail\arm\fre
IoTCorePShell arm 10.0.0.0 Test
PS C:\iot-adk-addonkit-master\Workspace>New-IoTWorkspace C:\MyWorkspace Contoso arm
New IoTWorkSpace available at C:\MyWorkspace for arm
Opening workspace : C:\MyWorkspace\IoTWorkspace.xml
Corekit found OK
Test certs installed
ADK_VERSION : 10.0.17763.1
IOTCORE_VER : 10.0.17763.107
BSP_VERSION : 10.0.0.0
ADDONKITVER : 6.0.181121.1730
HostOS Info : Microsoft Windows Server 2012 R2 Standard - 6.3.9600 - en-US
IOTWKSPACE  : C:\MyWorkspace
OEM_NAME    : Contoso
BSP_ARCH    : arm
BSPPKG_DIR  : C:\MyWorkspace\Build\arm\pkgs
MSPKG_DIR   : C:\Program Files (x86)\Windows Kits\10\MSPackages\Retail\arm\fre
Copying Registry.Version
Copying Custom.Cmd
Copying Provisioning.Auto
Copying OEM.Sample
Copying Device.SystemInformation
Copying DeviceLayout.GPT4GB
Copying DeviceLayout.GPT8GB-R
Copying DeviceLayout.MBR4GB
Copying DeviceLayout.MBR8GB-R
Workspace ready!
IoTCorePShell arm 10.0.0.0 Test
PS C:\MyWorkspace>
IoTCorePShell arm 10.0.0.0 Test
PS C:\MyWorkspace>Import-IoTOEMPackage Recovery.*
Copying Recovery.GPT-BCD
Copying Recovery.GPT-BcdEdit
Copying Recovery.MBR-BCD
Copying Recovery.MBR-BcdEdit
Copying Recovery.WinPE
IoTCorePShell arm 10.0.0.0 Test
PS C:\MyWorkspace>importbsp RPi2 C:\RPi_BSP.zip
Extracting BSP zip into the temporary directory C:\MyWorkspace\BSPDIR
Processing C:\MyWorkspace\BSPDIR\Rpi2
No .pkg.xml files found in C:\MyWorkspace\Source-arm\BSP\Rpi2.
BSP copy completed
IoTCorePShell arm 10.0.0.0 Test
PS C:\MyWorkspace>New-IoTCabPackage All
**Processing Device.SystemInformation.wm.xml
(PkgBldr.Common) : error : Failed to load schema resource stream Schemas.Schema.OnecorePackageInfo.xsd for input Schema\
OnecorePackageInfo.xsd.
Error: C:\MyWorkspace\Common\Packages\Device.SystemInformation\Device.SystemInformation.wm.xml pkggen failed**
False
IoTCorePShell arm 10.0.0.0 Test
PS C:\MyWorkspace>

parameshbabu, 2018-11-27:
Thanks @rtarta . We will check this on the Microsoft Windows Server 2012 R2 Standard - 6.3.9600 - en-US and get back to you.

rtarta, 2018-11-28:
@parameshbabu  hi 
yes both of them exists in this path on my machine 
**C:\Program Files (x86)\Windows Kits\10\tools\bin\i386**

saraclay, 2019-04-17:
If this is still an issue, I encourage you to post your question in our forums, where we have dedicated staff there to help you out. Thank you.

closed 2019-04-17 by saraclay