ms-iot/iot-adk-addonkit #294: Error: IoT Core kit ver not found in registry

alexismichellyc, 2018-11-29:
[image]
Installed everything from scratch, as mentionned in the IOT Core manufacturing guide.
As soon as I open the Powershell IoTCorePShell I get this error message : Error: IoT Core kit ver not found in registry. Not sure what to do here.

parameshbabu, 2018-11-29:
This occurs when you have not installed IoT Core packages (ARM version). If you have installed the amd64 version, then you can ignore this and when you do setenv amd64, proper status will show up.

I'll look to improve this message further.

parameshbabu, 2018-12-05:
with #295 you should see what arch packages are installed on the first launch.
Closing this.

closed 2018-12-05 by parameshbabu