ms-iot/iot-adk-addonkit #299: New-IoTFFUImage stagging stop with 85%

mayconsouza, 2019-01-03:
ADK_VERSION : 10.0.17763.1
IOTCORE_VER : 10.0.17763.1
BSP_VERSION : 10.0.0.0
ADDONKITVER : 6.0.181128.1909
HostOS Info : Microsoft Windows 10 Pro - 10.0.17134 - pt-BR
IOTWKSPACE  : D:\PlayTable\WindowsIOT\_imagePlaytable\Playtable\Playmove_1
OEM_NAME    : Playmove
BSP_ARCH    : amd64

It just closes PowerShell and no log is created. My log last lines:

ThreadId16432 INFO [00:02:16] Queueing package Microsoft-IoTUAP-IoTCoreUpdateOSSetting-Package~31bf3856ad364e35~amd64~~10.0.17763.1 for: install
ThreadId16432 INFO [00:02:16] Staging progress: 82%
ThreadId16432 INFO [00:02:17] Staging progress: 83%
ThreadId16432 INFO [00:02:19] Staging progress: 84%
ThreadId16432 INFO [00:02:20] Staging progress: 85%
ThreadId16432 INFO [00:02:21] Staging progress: 86%

parameshbabu, 2019-03-06:
Can you please try with the latest IoTCore kit? and preferably a clean build

saraclay, 2019-04-17:
Closing this as the original poster has been inactive. Please open a new issue if this is still something that needs to be addressed.

closed 2019-04-17 by saraclay