ms-iot/iot-adk-addonkit #300: IoTCorePShell.cmd does not properly set execution policy.

barronbarnett, 2019-01-04:
When attempting to start the IoTCore Power Shell I was confronted with a never ending list of security warnings.

`Security warning
Run only scripts that you trust. While scripts from the internet can be useful, this script can potentially harm your
computer. If you trust this script, use the Unblock-File cmdlet to allow the script to run without this warning
message. Do you want to run C:\IoTCoreBSP\iot-adk-addonkit\Tools\IoTCoreImaging\Classes\IoTProvisioningXML.ps1?
[D] Do not run  [R] Run once  [S] Suspend  [?] Help (default is "D"): R`

Changing the launch options in the IoTCorePshell.cmd was able to end this issue by changing the **ExecutionPolicy** from _Unrestricted_ to _Bypass_.

`powershell -Command "Start-Process 'powershell.exe' -ArgumentList '-noexit -ExecutionPolicy Bypass -Command \". %~dp0Tools\Launchshell.ps1\"' -Verb runAs"`

If desired I can create a branch and submit a PR.

barronbarnett, 2019-01-04:
An additional note, it appears that when creating a workspace the .cmd created also attempts to use -_ExecutionPolicy Unrestricted_, which results in the same error.

parameshbabu, 2019-01-07:
thanks for the suggestion. Will make the change in the next update shortly

PR #301 (2019-01-09): v6.0.190104.1215 - Support MSIX and fix executionpolicy bypass
Fix #300 
Support MSIX and MSIXBUNDLE extensions - Only modern apps with newer extensions supported.
closed 2019-01-09 by parameshbabu