ms-iot/iot-adk-addonkit #303: Is it possible to create an IoT CAB package from a driver that contains multiple INFs?

hollypang, 2019-01-17:
Hi, 

Let's say I have my RS5 driver binary package that contains 3 universal-compliant INF files:
(1) base.inf
(2) extend1.inf
(3) extend2.inf

base.inf references the other INFs via;
COPYINF = extend1.inf
COPY = extend2.inf

I downloaded the latest iot-adk-addonkit for RS5 and followed the instructions in the iot-adk-addonkit-master\Tools\README.md with regards to create a new workspace.  After issuing the **New-IoTCabPacking All** command, I noticed that driver.cab was created, but when I opened it, I noticed there were 2 instances of base.inf, and extend1.inf and extend2.inf were not there.

Is there a way to generate a package and cab with multiple universal-compliant INF files where one is the base INF for IoT Core?

Thanks,
Holly

parameshbabu, 2019-01-17:
Can you try adding the inf manually in the generated wm.xml file?
<drivers>
  <driver
      <inf source = “base.inf”/>
  </driver>
  <driver
      <inf source = “extend1.inf”/>
  </driver>
  <driver
      <inf source = “extend2.inf”/>
  </driver>
</drivers>

hollypang, 2019-01-18:
Hi Paramesh,

I added the inf files manually in the wm.xml and the generated .cab file did not look correct.  It had two copies of the base.inf in it, and the other 2 INFs were missing.  I will continue looking into this.

Thanks for your help.

-Holly

hollypang, 2019-01-23:
Hi,

I added the files that are related to the INF and that seemed to work:
<drivers>
  <driver
      <inf source = “base.inf”/>
      <files>
           <file
                source=".\base_source.sys"
                name="base_source.sys" />
             <file
                source=".\base_source.dll"
                name="base_source.dll" />
        </files>
  </driver>
  <driver
      <inf source = “extend1.inf”/>
      <files>
           <file
                source=".\extend1_source.sys"
                name="extend1_source.sys" />
             <file
                source=".\extend1_source.dll"
                name="extend1_source.dll" />
        </files>
  </driver>
  <driver
      <inf source = “extend2.inf”/>
      <files>
           <file
                source=".\extend2_source.sys"
                name="extend2_source.sys" />
             <file
                source=".\extend2_source.dll"
                name="extend2_source.dll" />
        </files>
  </driver>
</drivers>

closed 2019-01-23 by hollypang