ms-iot/iot-adk-addonkit #305: Dragonboard 410 Image Error on 1803

dgrothman, 2019-02-22:
Building 410C on 1803

Opening workspace : C:\IoT\PULSEGROUP_1\IoTWorkspace.xml
Corekit found OK
Test certs installed
ADK_VERSION : 10.0.17763.1
IOTCORE_VER : 10.0.17763.107
BSP_VERSION : 10.0.0.0
ADDONKITVER : 6.0.181121.1730
HostOS Info : Microsoft Windows 10 Pro - 10.0.17763 - en-US
IOTWKSPACE  : C:\IoT\PULSEGROUP_1
OEM_NAME    : PULSEGROUP
BSP_ARCH    : arm
BSPPKG_DIR  : C:\IoT\PULSEGROUP_1\Build\arm\pkgs
MSPKG_DIR   : C:\Program Files (x86)\Windows Kits\10\MSPackages\Retail\arm\fre
IoTCorePShell arm 10.0.0.0 Test
PS C:\IoT\PULSEGROUP_1>New-IoTFFUImage INSiGHT test
ADK_VERSION : 10.0.17763.1
IOTCORE_VER : 10.0.17763.107
BSP_VERSION : 10.0.0.0
ADDONKITVER : 6.0.181121.1730
HostOS Info : Microsoft Windows 10 Pro - 10.0.17763 - en-US
Validating product feature ids
Reading feature ids in C:\IoT\PULSEGROUP_1\Source-arm\BSP\QCDB410C\Packages\QCDB410CFM.xml
Reading feature ids in C:\IoT\PULSEGROUP_1\Source-arm\BSP\QCDB410C\Packages\QCDB410CTestFM.xml
Reading feature ids in C:\IoT\PULSEGROUP_1\Common\Packages\OEMCommonFM.xml
Reading feature ids in C:\IoT\PULSEGROUP_1\Source-arm\Packages\OEMFM.xml
Checking Microsoft features in OEMInput file..
Checking OEM features in OEMInput file..
Building product specific packages
Processing Registry.Version.wm.xml
Processing Custom.Cmd.wm.xml
Processing Provisioning.Auto.wm.xml
Processing Recovery.WinPE.wm.xml

Security warning
Run only scripts that you trust. While scripts from the internet can be useful, this script can potentially harm your
computer. If you trust this script, use the Unblock-File cmdlet to allow the script to run without this warning
message. Do you want to run C:\IoT\PULSEGROUP_1\Source-arm\BSP\QCDB410C\tools\NewFFUImage-Hook.ps1?
[D] Do not run  [R] Run once  [S] Suspend  [?] Help (default is "D"): r
Using custom SMBIOS settings
Processing Custom.SMBIOS.wm.xml
True
Building FM files..
Exporting OEM FM files..
Processing OEMFMList..
Exporting QCDB410C BSP FM files
Processing QCDB410CFMFileList.xml
Creating Image..
See C:\IoT\PULSEGROUP_1\Build\arm\INSiGHT_test.log for progress
This will take a while...
ThreadId5716 ERROR: Error : ImageCommon!BuildCompDB: Some Conditional Features have been defined and determined to apply but the corresponding Feature is missing:
ThreadId5716 ERROR:     Error : FeatureID=OEM_SMBIOS_DEFAULT FMID=QCDB410C
Build Completed. See C:\IoT\PULSEGROUP_1\Build\arm\INSiGHT\test\Flash.ffu
True
IoTCorePShell arm 10.0.0.0 Test

When I flash it to the device I just get a WIndows Boot Manager Error with message Windows Failed to start and down below
Status: 0xc000000d
Info: An Unexpected error has occurred.

I can successfully flash 1709 images no problem, I have multiple dragonboards to test on and I am experiencing this issue on all of them.

I have included the log file also. 
[INSiGHT_test.log](https://github.com/ms-iot/iot-adk-addonkit/files/2892422/INSiGHT_test.log)

parameshbabu, 2019-02-22:
I guess you are missing IOT_EFIESP feature id in your oeminput xml file. This is necessary to include the mandatory BCD packages.

FinitelyFailed, 2019-02-26:
I have the same issue on 1809, I tried adding IOT_EFIESP. which resulted in a conflict:

```
ThreadId23456 ERROR: [00:00:02] onecore\base\cbs\mobile\iuvalidator\packagevalidationrules.cpp, PackageValidationRules::Rule_DetectFileCollisions, line 726, Error  , Error - File collision detected, file '\boot\boot.sdi' found in packages 'Microsoft-OneCore-Boot-EFI-Devices-Test-Package~31bf3856ad364e35~arm~~' and 'Microsoft-OneCore-Boot-EFI-Production-Devices-Package~31bf3856ad364e35~arm~~'.
ThreadId23456 ERROR: [00:00:02] onecore\base\cbs\mobile\iuvalidator\packagevalidationrules.cpp, PackageValidationRules::Rule_DetectFileCollisions, line 726, Error  , Error - File collision detected, file '\EFI\Boot\bootarm.efi' found in packages 'Microsoft-OneCore-Boot-EFI-Devices-Test-Package~31bf3856ad364e35~arm~~' and 'Microsoft-OneCore-Boot-EFI-Production-Devices-Package~31bf3856ad364e35~arm~~'.
ThreadId23456 ERROR: [00:00:02] onecore\base\cbs\mobile\iuvalidator\packagevalidationrules.cpp, PackageValidationRules::Rule_DetectFileCollisions, line 726, Error  , Error - File collision detected, file '\EFI\Microsoft\Boot\boot.stl' found in packages 'Microsoft-OneCore-Boot-EFI-Devices-Test-Package~31bf3856ad364e35~arm~~' and 'Microsoft-OneCore-Boot-EFI-Production-Devices-Package~31bf3856ad364e35~arm~~'.
ThreadId23456 ERROR: [00:00:02] onecore\base\cbs\mobile\iuvalidator\packagevalidationrules.cpp, PackageValidationRules::Rule_DetectFileCollisions, line 726, Error  , Error - File collision detected, file '\EFI\Microsoft\Boot\bootmgfw.efi' found in packages 'Microsoft-OneCore-Boot-EFI-Devices-Test-Package~31bf3856ad364e35~arm~~' and 'Microsoft-OneCore-Boot-EFI-Production-Devices-Package~31bf3856ad364e35~arm~~'.
ThreadId23456 ERROR: [00:00:02] onecore\base\cbs\mobile\iuvalidator\packagevalidationrules.cpp, PackageValidationRules::Rule_DetectFileCollisions, line 726, Error  , Error - File collision detected, file '\EFI\Microsoft\Boot\bootstat.dat' found in packages 'Microsoft-OneCore-Boot-EFI-Devices-Test-Package~31bf3856ad364e35~arm~~' and 'Microsoft-OneCore-Boot-EFI-Production-Devices-Package~31bf3856ad364e35~arm~~'.
ThreadId23456 ERROR: [00:00:02] onecore\base\cbs\mobile\iuvalidator\packagevalidationrules.cpp, PackageValidationRules::Rule_DetectFileCollisions, line 726, Error  , Error - File collision detected, file '\EFI\Microsoft\Boot\kd_8003_5143.dll' found in packages 'Microsoft-OneCore-Boot-EFI-Devices-Test-Package~31bf3856ad364e35~arm~~' and 'Microsoft-OneCore-Boot-EFI-Production-Devices-Package~31bf3856ad364e35~arm~~'.
ThreadId23456 ERROR: [00:00:02] onecore\base\cbs\mobile\iuvalidator\packagevalidationrules.cpp, PackageValidationRules::Rule_DetectFileCollisions, line 726, Error  , Error - File collision detected, file '\EFI\Microsoft\Boot\kdstub.dll' found in packages 'Microsoft-OneCore-Boot-EFI-Devices-Test-Package~31bf3856ad364e35~arm~~' and 'Microsoft-OneCore-Boot-EFI-Production-Devices-Package~31bf3856ad364e35~arm~~'.
ThreadId23456 ERROR: [00:00:02] onecore\base\cbs\mobile\iuvalidator\packagevalidationrules.cpp, PackageValidationRules::Rule_DetectFileCollisions, line 726, Error  , Error - File collision detected, file '\EFI\Microsoft\Boot\mobilestartup.efi' found in packages 'Microsoft-OneCore-Boot-EFI-Devices-Test-Package~31bf3856ad364e35~arm~~' and 'Microsoft-OneCore-Boot-EFI-Production-Devices-Package~31bf3856ad364e35~arm~~'.
ThreadId23456 ERROR: [00:00:02] onecore\base\cbs\mobile\iuvalidator\packagevalidationrules.cpp, PackageValidationRules::Rule_DetectFileCollisions, line 726, Error  , Error - File collision detected, file '\EFI\Microsoft\Boot\winsipolicy.p7b' found in packages 'Microsoft-OneCore-Boot-EFI-Devices-Test-Package~31bf3856ad364e35~arm~~' and 'Microsoft-OneCore-Boot-EFI-Production-Devices-Package~31bf3856ad364e35~arm~~'.
ThreadId23456 ERROR: [00:00:02] onecore\base\cbs\mobile\iuvalidator\packagevalidationrules.cpp, PackageValidationRules::Rule_DetectFileCollisions, line 726, Error  , Error - File collision detected, file '\windows\winsxs\manifests\arm_microsoft-windows-b..bootmanager-efi-esp_31bf3856ad364e35_10.0.17763.107_none_c150495fb58804e2.manifest' found in packages 'Microsoft-OneCore-Boot-EFI-Devices-Test-Package~31bf3856ad364e35~arm~~' and 'Microsoft-OneCore-Boot-EFI-Production-Devices-Package~31bf3856ad364e35~arm~~'.
ThreadId23456 ERROR: [00:00:02] onecore\base\cbs\mobile\iuvalidator\packagevalidationrules.cpp, PackageValidationRules::Rule_DetectFileCollisions, line 726, Error  , Error - File collision detected, file '\windows\winsxs\manifests\arm_microsoft-windows-b..startup-bootsdi-efi_31bf3856ad364e35_10.0.17763.107_none_c96a2e7ce8d7f856.manifest' found in packages 'Microsoft-OneCore-Boot-EFI-Devices-Test-Package~31bf3856ad364e35~arm~~' and 'Microsoft-OneCore-Boot-EFI-Production-Devices-Package~31bf3856ad364e35~arm~~'.
ThreadId23456 ERROR: [00:00:03] onecore\base\cbs\mobile\updatedll\lib\updatemain.cpp, UpdateMain::ValidatePackages, line 258, Error  , Failed to calculate update order, error is 0x80188306
ThreadId23456 ERROR: [00:00:03] onecore\base\cbs\mobile\updatedll\lib\prepareupdate.cpp, UpdateMain::PrepareUpdate, line 177, Error  , Package verification failed, error is 0x80188306
ThreadId23456 ERROR: Imaging!UpdateImage: Failed call to UpdateDLL::PrepareUpdate with error code: -2145877242 (0x80188306)
Error: Build failed
```

parameshbabu, 2019-02-26:
I guess you are already using IOT_EFIESP_TEST. I suggest you try removing the IOT_EFIESP_TEST and use only IOT_EFIESP.

FinitelyFailed, 2019-03-05:
Using IOT_EFIESP and not IOT_EFIESP_TEST yields the same result as originally described.

parameshbabu, 2019-03-05:
Did you try a clean build? Lot of temp files might be reused causing this error again.

FinitelyFailed, 2019-03-11:
I've created a new product, changed the TestOEMInput.xml file so that it uses the IOT_EFIESP instead of IOT_EFIESP_TEST and than tried to build the image (buildimage product2 Test). Got the same error. I didn't find any commands for cleaning the build (or workspace?).

saraclay, 2019-04-17:
If this is still an issue, I encourage you to post your question in our forums, where we have dedicated staff there to help you out. Thank you.

closed 2019-04-17 by saraclay