ms-iot/iot-adk-addonkit #307: Disable Boot UX Progress Animation in Custom.BCD.xml does not work

aegiMt, 2019-03-05:
Hi. I am not able to hide the Boot Progress Animation Circle when building a new OS Image. I was following the steps mentoined in https://docs.microsoft.com/en-us/windows-hardware/manufacture/iot/oscustomizations, chapter BCD Settings. I adapted the Custom.BCD.xml file and added the feature to our image. 

Is the WellKnownTpye "Boot UX Progress Animation Disable" correct?

Please provide a list of WellKnownTypes.

Can we use this settings also for the Safe OS (Recovery.BCD.xml) to avoid the boot progress indicator when booting into Safe OS.

Many thanks

Andi

aegiMt, 2019-03-05:
I have attached our Custom.BCD.xml here:
[Custom.BCD.zip](https://github.com/ms-iot/iot-adk-addonkit/files/2930230/Custom.BCD.zip)

saraclay, 2019-04-17:
If this is still an issue, I encourage you to post your question in [our forums](https://social.msdn.microsoft.com/forums/en-US/home?forum=WindowsIoT), where we have dedicated staff there to help you out. Thank you.

closed 2019-04-17 by saraclay