ms-iot/iot-adk-addonkit #308: buildimage fails to validate cannot find FM file

ScottyJack, 2019-03-05:
I am trying to build an X64 IoT image and all steps are successful up until buildimage.  buildimage is throwing an access violation exception possibly while processing the OEMFMFileList.  The host machine is running Win 7 SP1.   Belwo is the OEMFMFilelist.xml and the buildimage output.  Any ideas on what the problem might be?

Thanks.

﻿<?xml version="1.0" encoding="utf-8"?>
<FMCollectionManifest xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xmlns:xsd="http://www.w3.org/2001/XMLSchema" xmlns="http://schemas.microsoft.com/embedded/2004/10/ImageUpdate" Product="IoTUAP">
    <IsBuildFeatureEnabled>true</IsBuildFeatureEnabled>
    <FMs>
        <FM ID="OEM" Path="$(FMDirectory)\OEMFM.xml" ReleaseType="Production" OwnerName="OEM_NAME" OwnerType="OEM" CPUType="AMD64" Critical="true" />
        <FM ID="OEMCOMMON" Path="$(FMDirectory)\OEMCOMMONFM.xml" ReleaseType="Production" OwnerName="OEM_NAME" OwnerType="OEM" CPUType="AMD64" Critical="true" />
    </FMs>
    <SupportedResolutions>
        <Resolution>1024x768</Resolution>
    </SupportedResolutions>
    <SupportedLanguages>
        <Language>en-US</Language>
    </SupportedLanguages>
    <SupportedLocales>
        <Locale>en-US</Locale>
    </SupportedLocales>
</FMCollectionManifest>

PS C:\iot-adk-addonkit-master\p6ws1>buildimage p6img1 Test
ADK_VERSION : 10.0.17763.1
IOTCORE_VER : 10.0.17763.107
BSP_VERSION : 10.0.0.0
ADDONKITVER : 6.0.190116.1218
HostOS Info : Microsoft Windows 7 Professional  - 6.1.7601 - en-US
Validating product feature ids
Reading feature ids in C:\iot-adk-addonkit-master\p6ws1\Source-x64\BSP\BYTx64\Pa
ckages\BYTx64FM.xml
Reading feature ids in C:\iot-adk-addonkit-master\p6ws1\Common\Packages\OEMCommo
nFM.xml
Reading feature ids in C:\iot-adk-addonkit-master\p6ws1\Source-x64\Packages\OEMF
M.xml
Checking Microsoft features in OEMInput file..
Warning: IOT_APPLICATIONS is not defined
Warning: IOT_DISABLE_UMCI is not defined
Warning: IOT_NETCMD is not defined
Warning: PRODUCTION_CORE is not defined
Checking OEM features in OEMInput file..
Building product specific packages
Processing Registry.Version.wm.xml
Processing Custom.Cmd.wm.xml
Processing Provisioning.Auto.wm.xml
Processing Recovery.WinPE.wm.xml
Building FM files..
Exporting OEM FM files..
Processing OEMFMList..

Unhandled Exception: System.AccessViolationException: Attempted to read or write
 protected memory. This is often an indication that other memory is corrupt.
   at Microsoft.WindowsPhone.ImageUpdate.PkgCommon.PkgConvertDSM.ConvertListOfSP
KGs(String[] packagesList, UInt32 cPackageList, CONVERTDSM_PARAMETERS& pParams)
   at Microsoft.WindowsPhone.ImageUpdate.PkgCommon.PkgConvertDSM.ConvertPackages
ToCBS(CONVERTDSM_PARAMETERS_FLAGS Flags, List`1 packageList, String outputDir)
   at Microsoft.WindowsPhone.ImageUpdate.FeatureMerger.FeatureMerger.<>c__Displa
yClass53_0.<GenerateCBSPackages>b__0(String spkgFile)
   at System.Threading.Tasks.Parallel.<>c__DisplayClass31_0`2.<ForEachWorker>b__
0(Int32 i)
   at System.Threading.Tasks.Parallel.<>c__DisplayClass17_0`1.<ForWorker>b__1()
   at System.Threading.Tasks.Task.InnerInvoke()
   at System.Threading.Tasks.Task.InnerInvokeWithArg(Task childTask)
   at System.Threading.Tasks.Task.<>c__DisplayClass176_0.<ExecuteSelfReplicating
>b__0(Object )
   at System.Threading.Tasks.Task.ExecuteSelfReplicating(Task root)
   at System.Threading.Tasks.Task.Execute()
   at System.Threading.Tasks.Task.ExecutionContextCallback(Object obj)
   at System.Threading.ExecutionContext.RunInternal(ExecutionContext executionCo
ntext, ContextCallback callback, Object state, Boolean preserveSyncCtx)
[... 198 lines cut ...]
1 source, ParallelOptions parallelOptions, Action`1 body, Action`2 bodyWithState
, Action`3 bodyWithStateAndIndex, Func`4 bodyWithStateAndLocal, Func`5 bodyWithE
verything, Func`1 localInit, Action`1 localFinally)
   at System.Threading.Tasks.Parallel.ForEach[TSource](IEnumerable`1 source, Act
ion`1 body)
   at Microsoft.WindowsPhone.ImageUpdate.FeatureMerger.FeatureMerger.GenerateCBS
Packages(List`1 packageList, CONVERTDSM_PARAMETERS_FLAGS flags)
   at Microsoft.WindowsPhone.ImageUpdate.FeatureMerger.FeatureMerger.GenerateCBS
PackagesWithTestSigning(List`1 packageList)
   at Microsoft.WindowsPhone.ImageUpdate.FeatureMerger.FeatureMerger.MergeOneFM(
FMCollectionItem fmItem)
   at Microsoft.WindowsPhone.ImageUpdate.FeatureMerger.FeatureMerger.MergeFeatur
es(String inputFile, String outputPackageDir, String outputFMDir, String version
Str, String inputFMDir, String critical, String packagePathReplacement, String v
ariablesStr, String fmID, String ownerTypeStr, String ownerName, String language
s, String resolutions, Boolean compress, Boolean convertToCBS, Boolean increment
al, Boolean doPlatformManifest)
   at Microsoft.WindowsPhone.ImageUpdate.FeatureMerger.FeatureMerger.Main(String
[] args)
Error: New-IoTFIPPackage failed to process BSPFMList. See C:\iot-adk-addonkit-ma
ster\p6ws1\Build\amd64\FIPPackage_BYTx64.log
Error: Failed to create FIP packages
False
IoTCorePShell amd64 10.0.0.0 Test
PS C:\iot-adk-addonkit-master\p6ws1>

ScottyJack, 2019-03-05:
I should also add that I am running powershell 6.1.3 which I updated from the original version on the host machine which as I recall now might have been 2.0.

parameshbabu, 2019-03-05:
I recommend to use a Windows 10 machine for this. I see you are running on a Windows 7 machine

ScottyJack, 2019-03-07:
Ok, I will give that a try.

closed 2019-04-17 by saraclay