ms-iot/iot-adk-addonkit #309: How to represent the custom destinationDir?

moyanming, 2019-03-07:
Hi,
According to the steps [here](https://docs.microsoft.com/en-us/windows-hardware/manufacture/iot/add-a-registry-setting-to-an-image#build-a-package-for-your-test-files), how to represent the destinationDir to the "c:\data\users\MySettings"?

If I use the "c:\data\users\MySettings", I got the error:
 Processing Registry.FilesAndRegKeys.wm.xml
(14,11): error “destinationDir”特性无效 - 根据数据类型“urn:Microsoft.CompPlat/ManifestSchema.v1.00:DevicePathType”，值“c:\data\users\MySettings”无效 - 模式约束失败。
(PkgBldr.Common) : error : PkgBldrCommon schema validation failed
Error : Failed to create package. See C:\IoT-ADK-AddonKit\Build\arm\pkgs\logs\Registry.FilesAndRegKeys.wm.log

parameshbabu, 2019-03-07:
Try using $(runtime.data). See C:\Program Files (x86)\Windows Kits\10\Tools\bin\i386\pkggen.cfg.xml

moyanming, 2019-03-08:
Hi @parameshbabu , 
I am sorry for give you the wrong website. I use the 17134 [here](https://github.com/ms-iot/iot-adk-addonkit/blob/17134/Docs/add-a-registry-setting-to-an-image.md#add-sample-files-to-the-package). After I use the $(runtime.data), I still have the wrong message.
Any ideas? 
Thanks.
`Creating Registry.FilesAndRegKeys.wm.xml Package with version 10.0.0.0 for DigitalSign
Microsoft (C) PkgGen 10.0.10011.16384

--Stack Trace--
   在 Microsoft.CompPlat.PkgBldr.Base.MacroResolver.MatchEvaluator.Evaluate(Match match)
   在 System.Text.RegularExpressions.RegexReplacement.Replace(MatchEvaluator evaluator, Regex regex, String input, Int32 count, Int32 startat)
   在 System.Text.RegularExpressions.Regex.Replace(String input, MatchEvaluator evaluator, Int32 count, Int32 startat)
   在 System.Text.RegularExpressions.Regex.Replace(String input, MatchEvaluator evaluator)
   在 Microsoft.CompPlat.PkgBldr.Base.MacroResolver.Resolve(String input, MacroResolveOptions option)
   在 Microsoft.CompPlat.PkgBldr.Base.MacroResolver.Resolve(String input)
   在 Microsoft.CompPlat.PkgBldr.Plugins.WmToCsi.File.ConvertEntries(XElement ToCsi, Dictionary`2 plugins, Config enviorn, XElement FromWm)
   在 Microsoft.CompPlat.PkgBldr.Base.PkgPlugin.ConvertEntries(XElement parent, Dictionary`2 plugins, Config enviorn, XElement component)
   在 Microsoft.CompPlat.PkgBldr.Plugins.WmToCsi.Files.ConvertEntries(XElement ToCsi, Dictionary`2 plugins, Config enviorn, XElement fromWm)
   在 Microsoft.CompPlat.PkgBldr.Base.PkgPlugin.ConvertEntries(XElement parent, Dictionary`2 plugins, Config enviorn, XElement component)
   在 Microsoft.CompPlat.PkgBldr.Plugins.WmToCsi.Identity.ConvertEntries(XElement ToCsi, Dictionary`2 plugins, Config enviorn, XElement FromWm)
   在 Microsoft.CompPlat.PkgBldr.Program.BuildPackage(Config config, MacroResolver commandLineMacros)
   在 Microsoft.CompPlat.PkgBldr.Program.Main(String[] args)
--End Stack Trace--
Package creation failed with error -1
`

parameshbabu, 2019-03-08:
You can try $(runtime.bootDrive)\data . In general it is not recommended to have packages target data partition. If these are settings that you need to preserve, its best to create an directory in the root say $(runtime.bootDrive)\OEM.

moyanming, 2019-03-09:
Hi @parameshbabu ,
I don't want to occupy the space of the MainOS (I think the $(runtime.bootDrive)\OEM is in the MainOS partition.?). So I want to store the data in the Data partition.
BTW, does the "c:\data" is link to the data partition?

saraclay, 2019-04-17:
If this is still an issue, I encourage you to post your question in [our forums](https://social.msdn.microsoft.com/forums/en-US/home?forum=WindowsIoT), where we have dedicated staff there to help you out. Thank you.

closed 2019-04-17 by saraclay