ms-iot/iot-adk-addonkit #310: How to confirm the partition layout after program the custom FFU on the DB410C?

moyanming, 2019-03-09:
Hi,
I want to add all the Speech data feature, those speech data may occupy more space of the EMMC, so,
https://github.com/ms-iot/iot-adk-addonkit/blob/c10904278e33070939b0e1b8391f2b2f17d9820c/Source-arm/BSP/QCDB410C/Packages/QCDB410C.DeviceLayout-R/DeviceLayout.xml#L240
1. Do I need to increase the MinFreeSectors  of the MainOS partition or other partition of the QCDB410C.DeviceLayout?
2. Which data will be store in the MinFreeSectors of the MainOS partition?
3. I want to delete the recovery partition (I need more user data space), is it right that just delete the following partition?
    
    <Partition>
      <Name>MMOS</Name>
      <FileSystem>FAT32</FileSystem>
      <TotalSectors>4096000</TotalSectors>
      
      <Type>{ebd0a0a2-b9e5-4433-87c0-68b6b72699c7}</Type>
    </Partition>
4. How to confirm the partition layout after program my custom FFU on the DB410C?
Is there diskpart or other command can be used?

Best regards.

moyanming, 2019-03-29:
Get-PSDrive
can list the following:
[192.168.3.3]: PS C:\Data\Users\Administrator\Documents> get-psdrive

Name           Used (GB)     Free (GB) Provider      Root                                               CurrentLocation
----           ---------     --------- --------      ----                                               ---------------
Alias                                  Alias
C                   0.91          0.50 FileSystem    C:\                                 ...ers\Administrator\Documents
Cert                                   Certificate   \
D                   0.00          0.01 FileSystem    D:\
Env                                    Environment
Function                               Function
HKCU                                   Registry      HKEY_CURRENT_USER
HKLM                                   Registry      HKEY_LOCAL_MACHINE
U                   0.48          5.22 FileSystem    U:\
Variable                               Variable
WSMan                                  WSMan

parameshbabu, 2019-04-04:
1. You dont need to increase the MinFreeSectors as such as this is the free space for MainOS added after the actual size of the MainOS ( so when you add more features, the actual size of the MainOS increases).
2. The current value is good enough and no need to change.
3. Yes, you can remove that Partition definition if you dont need MMOS. Note that you should also remove the recovery related features from the OEM Input xml file (RECOVERY_BCD etc). Also with this change, you will not be able to recover the device in case of any failures during update.
4. You can mount the FFU on the pc using ffum command and see the partition layout.

closed 2019-04-17 by saraclay