ms-iot/iot-adk-addonkit #311: IOT_UAP_OOBE in Headless mode

moyanming, 2019-03-11:
Hi,
The[ IOT_UPA_OOBE](https://docs.microsoft.com/en-us/windows-hardware/manufacture/iot/iot-core-feature-list#retail-features-defined-by-microsoft) is the required feature in all images. But this feature is a foreground app, I want use this IOT_UAP_OOBE for the APPX installation by use the Windows Device Portal. How do this in a headless mode image? What happend if I not include this feature?
[image]

Best regards.

saraclay, 2019-04-17:
@parameshbabu is this something you might know?

parameshbabu, 2019-05-13:
IOT_UAP_OOBE should be included in the image as this is the first app that is launched by the OS before the user defined app is installed. Also this app is run when you uninstall an app and install another app.
Even for the headless mode, do include this app by default.

moyanming, 2019-05-14:
Hi @parameshbabu ,
If the custom image includes the “IOT_HEADLESS_CONFIGURATION” feature and this means the foreground app will not launch. How to install and uninstall by the headless mode OS as the IOT_UAP_OOBE will not launch?
Does the “IOT_HEADLESS_CONFIGURATION” feature is the same as the command: "setbootoption.exe headless"?

parameshbabu, 2019-05-14:
I was incorrect in my earlier response. You can skip IoT_UAP_OOBE in the headless configuration when you include IOT_HEADLESS_CONFIGURATION. This puts the device in headless mode by default. 
This is same as doing setbootoption headless and rebooting the device. I will update the documentation to reflect this. There is no harm including this though as it will be there installed and will not launch.

closed 2020-03-06 by parameshbabu