ms-iot/iot-adk-addonkit #312: ThreadId4728 ERROR

moyanming, 2019-03-24:
Hi
A clean OS v17763. DB410c bsp v2121.
There is an error occur in the "new-iotffuimage xxx test" command. But the FFU is still build and I can program and run this FFU image successfully.

ThreadId4728 EROR: Error : ImageCommon!BuildCompDB: Some Conditional Features have been defined and determined to apply but the conrresponding Feature is missing: ThreadId4728 Error:
Error: FeatureID=OME_SMBIOS_DEFAULT FMID=QCDB410C

How to fix this problem?
[image]

saraclay, 2019-04-17:
@moyanming, for these questions, please post them in [our forums](https://social.msdn.microsoft.com/forums/en-US/home?forum=WindowsIoT) where we have dedicated staff to help.

luyaok, 2019-05-10:
I can reproduce this issue too. Follow [this issue](https://github.com/ms-iot/iot-adk-addonkit/issues/305) replace the IOT_EFIESP_TEST with IOT_EFIESP but still same result.

**This is my OEMInput.xml.**

`<?xml version="1.0" encoding="utf-8"?>
<OEMInput xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xmlns:xsd="http://www.w3.org/2001/XMLSchema" xmlns="http://schemas.microsoft.com/embedded/2004/10/ImageUpdate">
  <Description>Windows 10 IoT Core 8016sbc Test FFU</Description>
  <SOC>QC8016</SOC>
  <SV>Qualcomm</SV>
  <Device>8016sbc</Device>
  <ReleaseType>Test</ReleaseType>
  <BuildType>fre</BuildType>
  <SupportedLanguages>
    <UserInterface>
      <Language>en-us</Language>
    </UserInterface>
    <Keyboard>
      <Language>en-us</Language>
    </Keyboard>
    <Speech>
      <Language>en-us</Language>
    </Speech>
  </SupportedLanguages>
  <BootUILanguage>en-us</BootUILanguage>
  <BootLocale>en-us</BootLocale>
  <Resolutions>
    <Resolution>1024x768</Resolution>
  </Resolutions>
  <AdditionalFMs>
    <AdditionalFM>C:\MyWorkspace\DragonBoard410\Build\arm\MergedFMs\QCDB410CFM.xml</AdditionalFM>
    <AdditionalFM>C:\MyWorkspace\DragonBoard410\Build\arm\MergedFMs\QCDB410CTestFM.xml</AdditionalFM>
    <AdditionalFM>C:\MyWorkspace\DragonBoard410\Build\arm\MergedFMs\OEMCommonFM.xml</AdditionalFM>
    <AdditionalFM>C:\MyWorkspace\DragonBoard410\Build\arm\MergedFMs\OEMFM.xml</AdditionalFM>
    <AdditionalFM>C:\Program Files (x86)\Windows Kits\10\FMFiles\arm\IoTUAPNonProductionPartnerShareFM.xml</AdditionalFM>
  </AdditionalFMs>
  <Features>
    <Microsoft>
      <Feature>IOT_UAP_OOBE</Feature>
      <Feature>IOT_CRT140</Feature>
      <Feature>IOT_UNIFIED_WRITE_FILTER</Feature>
      <Feature>IOT_USBFN_CLASS_EXTENSION</Feature>
      <Feature>IOT_POWERSHELL</Feature>
      <Feature>IOT_ONBOARDING_APP</Feature>
      <Feature>IOT_APP_TOOLKIT</Feature>
      <Feature>IOT_FFU_FLASHMODE</Feature>
      <Feature>IOT_MTP</Feature>
      <Feature>IOT_MIRACAST_RX_APP</Feature>
      <Feature>IOT_WEBB_EXTN</Feature>
      <Feature>IOT_BLUETOOTH_A2DP_SINK</Feature>
      <Feature>IOT_BLUETOOTH_A2DP_SOURCE</Feature>
      <Feature>IOT_BLUETOOTH_HFP_AUDIOGATEWAY</Feature>
      <Feature>IOT_NARRATOR</Feature>
      <Feature>IOT_OCR_EN_US</Feature>
      <Feature>IOT_POWER_SETTINGS</Feature>
      <Feature>IOT_EFIESP_BCD</Feature>
      <Feature>IOT_SHELL_HOTKEY_SUPPORT</Feature>
      <Feature>IOT_SHELL_ONSCREEN_KEYBOARD</Feature>
      <Feature>IOT_SHELL_ONSCREEN_KEYBOARD_FOLLOWFOCUS</Feature>
      <Feature>IOT_DISABLEBASICDISPLAYFALLBACK</Feature>
      <Feature>IOT_CRASHCONTROL_SETTINGS</Feature>
      <Feature>IOT_SSH</Feature>
      <Feature>IOT_SIREP</Feature>
      <Feature>IOT_TOOLKIT</Feature>
      <Feature>IOT_NANORDPSERVER</Feature>
      <Feature>IOT_BERTHA</Feature>
      <Feature>IOT_FTSER2K_MAKERDRIVER</Feature>
      <Feature>IOT_CP210x_MAKERDRIVER</Feature>
      <Feature>IOT_CONTAINERS</Feature>
      <Feature>IOT_ENABLE_TESTSIGNING</Feature>
      <Feature>IOT_KDSERIAL_SETTINGS</Feature>
      <Feature>IOT_WDTF</Feature>
      <Feature>IOT_DIRECTX_TOOLS</Feature>
      <Feature>IOT_UMDFDBG_SETTINGS</Feature>
      <Feature>IOT_EFIESP</Feature>
    </Microsoft>
    <OEM>
      <Feature>QC_UEFI_TEST</Feature>
      <Feature>SBC</Feature>
      <Feature>CUSTOM_CMD</Feature>
      <Feature>PROV_AUTO</Feature>
      <Feature>CUSTOM_SMBIOS</Feature>
    </OEM>
  </Features>
  <AppXOptionalPackages />
  <AdditionalSKUs />
  <InternalOptionalFeatures />
  <ProductionOptionalFeatures />
  <MSOptionalFeatures />
  <PackageFiles />
  <Product>Windows 10 IoT Core</Product>
</OEMInput>`

**This is error message in Flash.ImageApp.log.**

`{5/10/2019 2:06:44 PM} Error: Error : ImageCommon!BuildCompDB: Some Conditional Features have been defined and determined to apply but the corresponding Feature is missing:
{5/10/2019 2:06:44 PM} Error: 	Error : FeatureID=OEM_SMBIOS_DEFAULT FMID=QCDB410C
{5/10/2019 2:06:59 PM}  Imaging: Writing BSP CompDB to '\\?\C:\MYWORK~1\DRAGON~1\Build\arm\ProductA\Test\Flash.BSPDB.xml' ...
{5/10/2019 2:06:59 PM}  Imaging: Writing Device CompDB to '\\?\C:\MYWORK~1\DRAGON~1\Build\arm\ProductA\Test\Flash.DeviceDB.xml' ...
{5/10/2019 2:07:00 PM}  Imaging: Finalizing the FFU image...`

parameshbabu, 2019-05-10:
This is a false alarm and can be ignored. I will get this suppressed. You can get rid of this by removing this line in the QCDB401CFM.xml file.
```
      <ConditionalFeature FeatureID="OEM_SMBIOS_DEFAULT" UpdateAction="NoUpdate" />
```

closed 2019-05-13 by moyanming