ms-iot/iot-adk-addonkit #313: Retail Build : How to open specific COM & firewall ports

majidshaffi, 2019-03-25:
Hi All,

I am working on the retail build and our UWP app talks to an inverter and battery on different comm ports. It talks to battery on COM Port 0 and to inverter on COM Port 2.

By default I think, COM Port 0 is kept open and for this reason the app is able to talk to battery but we need to open COM Port 2 so that it can talk to inverter as well.

I checked that if we add **IoT_ToolKit** feature in the retail image it does open the ports but it open all the ports and it is also a test feature which should not be added in Retail.

Could anyone tell where are the COM Port information is stored and if we can execute a PowerShell script to enable COM Port 2?

Thanks

majidshaffi, 2019-03-27:
@parameshbabu @imingc ,

Could you please guide us on this matter please?

majidshaffi, 2019-04-02:
Hi @parameshbabu ,

Could you please provide some insights on this?

saraclay, 2019-04-17:
@majidshaffi I encourage you to post your question in [our forums](https://social.msdn.microsoft.com/forums/en-US/home?forum=WindowsIoT), where we have dedicated staff there to help you out.