ms-iot/iot-adk-addonkit #314: Cannot build FFU image- suspect that the issue is due to workspace defaulting to ARM64

sunnenproducts, 2019-04-02:
I cannot progress past lab 1a : https://docs.microsoft.com/en-us/windows-hardware/manufacture/iot/create-a-basic-image  

The error says: "fatal error PKG: 2019-04-02 09:38:56, Error                 SXS    Failed to extract C:\Program Files (x86)\Windows Kits\10\MSPackages\Retail\AMD64\fre\Microsoft-IoTUAP-Product-Package~31bf3856ad364e35~AMD64~~.cab: HRESULT_FROM_WIN32(ERROR_ACCESS_DENIED)"

You will notice that this  error refers to AMD64

I notice that Windows has updated to 1809 and iot-adk-addonkit is for 1803 - could something have changed?  Is there now a bug???

For some reason, when I create a workspace, the BSP_ARCH is AMD64,  as well as BSPPKG_DIR, & MSPKG_DIR.  Please see below my entire Powershell interaction.

PS C:\0-iot-adk-addonkit-master\Workspace>new-iotworkspace c:\myworkspace mycompany x64
New IoTWorkSpace available at c:\myworkspace for x64
Opening workspace : C:\myworkspace\IoTWorkspace.xml
Corekit found OK
ADK_VERSION : 10.0.17763.1
IOTCORE_VER : 10.0.17763.253
BSP_VERSION : 10.0.0.0
ADDONKITVER : 6.0.190116.1218
HostOS Info : Microsoft Windows 10 Pro - 10.0.17763 - en-US
IOTWKSPACE  : C:\myworkspace
OEM_NAME    : mycompany
BSP_ARCH    : amd64
BSPPKG_DIR  : C:\myworkspace\Build\amd64\pkgs
MSPKG_DIR   : C:\Program Files (x86)\Windows Kits\10\MSPackages\Retail\amd64\fre
Copying Registry.Version
Copying Custom.Cmd
Copying Provisioning.Auto
Copying OEM.Sample
Copying Device.SystemInformation
Copying DeviceLayout.GPT4GB
Copying DeviceLayout.GPT8GB-R
Copying DeviceLayout.MBR4GB
Copying DeviceLayout.MBR8GB-R
Workspace ready!
IoTCorePShell amd64 10.0.0.0 Test
PS C:\myworkspace>Import-IoTOEMPackage *
Copying Appx.Certs
Copying Custom.BCD
Copying Custom.OOBEApp
Copying Custom.Settings
Warning: C:\myworkspace\Common\Packages\Device.SystemInformation already exist.
Warning: C:\myworkspace\Common\Packages\DeviceLayout.GPT4GB already exist.
Warning: C:\myworkspace\Common\Packages\DeviceLayout.GPT8GB-R already exist.
Warning: C:\myworkspace\Common\Packages\DeviceLayout.MBR4GB already exist.
Warning: C:\myworkspace\Common\Packages\DeviceLayout.MBR8GB-R already exist.
Copying Recovery.GPT-BCD
Copying Recovery.GPT-BcdEdit
Copying Recovery.MBR-BCD
Copying Recovery.MBR-BcdEdit
Warning: C:\myworkspace\Common\Packages\Registry.Version already exist.
Copying Security.BitLocker
Copying Security.DeviceGuard
Copying Security.DeviceGuardTest
Copying Security.SecureBoot
Copying Security.SecureBootTest
Copying Settings.HotKey
Copying Appx.IoTCoreDefaultApp
Copying Appx.IoTOnboardingTask
Copying AzureDM.Services
Warning: C:\myworkspace\Source-x64\Packages\OEM.Sample already exist.
Warning: C:\myworkspace\Common\ProdPackages\Custom.Cmd already exist.
Warning: C:\myworkspace\Common\ProdPackages\Provisioning.Auto already exist.
Copying Recovery.WinPE
IoTCorePShell amd64 10.0.0.0 Test
PS C:\myworkspace>import-iotbsp BSWx64 C:\0_BSP_UP2\Source-x64\BSP
Processing C:\0_BSP_UP2\Source-x64\BSP\BSWx64
Warning:   Replacing Intel.BSW64.DeviceLayout with %OEM_NAME%.BSW64.DeviceLayout
Warning:   Replacing Intel.BSW64.GFX with %OEM_NAME%.BSW64.GFX
Warning:   Replacing Intel.BSW64.OEMDevicePlatform with %OEM_NAME%.BSW64.OEMDevicePlatform
BSP copy completed
IoTCorePShell amd64 10.0.0.0 Test
PS C:\myworkspace>new-iotcabpackage all
Processing Appx.Certs.wm.xml
Processing Custom.BCD.wm.xml
Processing Custom.OOBEApp.wm.xml
Processing Custom.Settings.wm.xml
Processing Device.SystemInformation.wm.xml
Processing DeviceLayout.GPT4GB.wm.xml
Processing DeviceLayout.GPT8GB-R.wm.xml
Processing DeviceLayout.MBR4GB.wm.xml
Processing DeviceLayout.MBR8GB-R.wm.xml
Processing Recovery.GPT-BCD.wm.xml
Processing Recovery.GPT-BcdEdit.wm.xml
Processing Recovery.MBR-BCD.wm.xml
Processing Recovery.MBR-BcdEdit.wm.xml
Processing Registry.Version.wm.xml
Processing Security.Bitlocker.wm.xml
Processing Security.DeviceGuard.wm.xml
Processing Security.DeviceGuardTest.wm.xml
Processing Security.SecureBoot.wm.xml
Processing Security.SecureBootTest.wm.xml
Processing Settings.HotKey.wm.xml
Processing Appx.IoTCoreDefaultApp.wm.xml
Processing Appx.IoTOnboardingTask.wm.xml
Processing AzureDM.Services.wm.xml
Processing OEM.Sample.wm.xml
Processing BSW64.DeviceLayout.wm.xml
Processing BSW64.GFX.wm.xml
Processing BSW64.GPIO.wm.xml
Processing BSW64.I2C.wm.xml
Processing BSW64.OEMDevicePlatform.wm.xml
Processing BSW64.SystemInformation.wm.xml
Processing BSW64.UART.wm.xml
True
IoTCorePShell amd64 10.0.0.0 Test
PS C:\myworkspace>add-iotproduct pk BSWx64

cmdlet Add-IoTProduct at command pipeline position 1
Supply values for the following parameters:
OemName: Sunnen
FamilyName: PGGAUGE
SkuNumber: 800
BaseboardManufacturer: UP2
BaseboardProduct: 2019
Creating pk Product with BSP BSWx64
Creating C:\myworkspace\Source-x64\Products\pk\prov\customizations.xml...
Creating C:\myworkspace\Source-x64\Products\pk\pkSettings.xml...
Product Config file : C:\myworkspace\Source-x64\Products\pk\pkSettings.xml
DeviceInventory file : C:\myworkspace\Source-x64\Products\pk\IoTDeviceModel_pk.xml
OS Version           : 10.0.17763.253
BSP Version          : 10.0.0.0
DeviceInventory created
IoTCorePShell amd64 10.0.0.0 Test
PS C:\myworkspace>New-IoTFFUImage pk test
ADK_VERSION : 10.0.17763.1
IOTCORE_VER : 10.0.17763.253
BSP_VERSION : 10.0.0.0
ADDONKITVER : 6.0.190116.1218
HostOS Info : Microsoft Windows 10 Pro - 10.0.17763 - en-US
Validating product feature ids
Reading feature ids in C:\myworkspace\Source-x64\BSP\BSWx64\Packages\BSWx64FM.XML
Reading feature ids in C:\myworkspace\Common\Packages\OEMCommonFM.xml
Reading feature ids in C:\myworkspace\Source-x64\Packages\OEMFM.xml
Checking Microsoft features in OEMInput file..
Warning: IOT_APPLICATIONS is not defined
Warning: IOT_CORTANA is not defined
Warning: IOT_DISABLE_UMCI is not defined
Warning: IOT_GENERIC_POP is not defined
Warning: IOT_NETCMD is not defined
Checking OEM features in OEMInput file..
Building product specific packages
Processing Registry.Version.wm.xml
Processing Custom.Cmd.wm.xml
Processing Provisioning.Auto.wm.xml
Processing Recovery.WinPE.wm.xml
Building FM files..
Exporting OEM FM files..
Processing OEMFMList..
Exporting BSWx64 BSP FM files
Processing BSWx64FMFileList.xml
Creating Image..
See C:\myworkspace\Build\amd64\pk_test.log for progress
This will take a while...
ThreadId1484 ERROR: [00:02:31] onecore\base\cbs\mobile\iucore\iupartition.cpp, IUPartition::ProcessIndividualCbsPackages, line 911, Error  , Failed to finalize client, error is 0x8018830B
ThreadId1484 ERROR: [00:02:31] onecore\base\cbs\mobile\iucore\iupartition.cpp, IUPartition::ProcessAllCbsPackages, line 1245, Error  , Failed to process individual packages while Staging, error is 0x8018830B
ThreadId1484 ERROR: [00:02:31] onecore\base\cbs\mobile\iucore\iupartition.cpp, IUPartition::PrepareFiles, line 1422, Error  , Failed to process packages for partition MainOS, error is 0x8018830B
ThreadId1484 ERROR: [00:02:31] onecore\base\cbs\mobile\updatedll\lib\prepareupdate.cpp, UpdateMain::PrepareFiles, line 362, Error  , Failed to prepare files for partition: MainOS, error is 0x8018830B,
ThreadId1484 ERROR: [00:02:31] onecore\base\cbs\mobile\updatedll\lib\prepareupdate.cpp, UpdateMain::PrepareUpdate, line 191, Error  , File preparation failed, error is 0x8018830B
ThreadId1484 ERROR: Imaging!UpdateImage: Failed call to UpdateDLL::PrepareUpdate with error code: -2145877237 (0x8018830B)
Error: Build failed

parameshbabu, 2019-04-04:
x64 is same as AMD64.  so that is not an issue here.  Is the amd64 version of the IoTCore packages installed on this machine?

sunnenproducts, 2019-04-09:
Yes- I installed all architectures.

closed 2019-04-10 by sunnenproducts

timcarr, 2020-01-22:
Why is this closed? @sunnenproducts  did you find a solution?