ms-iot/iot-adk-addonkit #319: Windows IOT Intel - BYTx64FM.xml not found

ZejotnA, 2019-05-31:
Hello, I have to build an Windows IOT image for a 64-bit processor. When i run the "buildimage" command, i have this error : 
[image]

I tried to execute it too on x86 but there is the same error.
The BYTx64FM.xml is not found but he is present in the folder...

[image]

I have followed instruction of this website : https://sibeeshpassion.com/raspberry-pi-sd-card-provisioning-with-windows-10-iot-core-create-ffu-image-and-flash/#prerequisites

But with Intel BSP : https://downloadcenter.intel.com/download/25618/Intel-Embedded-Drivers-for-Microsoft-Windows-10-IoT-Core-32-bit-and-64-bit-MR1?product=29035

Could you help me ?

parameshbabu, 2019-05-31:
Hi,

Can you follow the instructions for Intel BSP here
https://docs.microsoft.com/en-us/windows-hardware/manufacture/iot/bsphardware#intel-bsps

Regards,
Paramesh

[image]<https://nam06.safelinks.protection.outlook.com/?url=https%3A%2F%2Fuser-images.githubusercontent.com%2F51236265%2F58707751-0ef90c80-83b6-11e9-8a1c-e61143f8c39f.png&data=02%7C01%7Cpabab%40microsoft.com%7C32c6cb4de7ec41c6087c08d6e5ca8fc0%7C72f988bf86f141af91ab2d7cd011db47%7C1%7C0%7C636949055489767394&sdata=oDDQO%2F%2FstT2hLZUB7DZjgo9Fb5GqGYCwxcvs4A6Ftug%3D&reserved=0>

[image]<https://nam06.safelinks.protection.outlook.com/?url=https%3A%2F%2Fuser-images.githubusercontent.com%2F51236265%2F58707947-a1011500-83b6-11e9-9df8-670faf2c6cc6.png&data=02%7C01%7Cpabab%40microsoft.com%7C32c6cb4de7ec41c6087c08d6e5ca8fc0%7C72f988bf86f141af91ab2d7cd011db47%7C1%7C0%7C636949055489767394&sdata=d0H3wqVKoyv3jUEUn1FWCpIU1AwKsjnm3SVwuBOzHwY%3D&reserved=0>

ZejotnA, 2019-05-31:
Thank's a lot !
It works! 
Have a good day!

closed 2019-05-31 by ZejotnA