ms-iot/iot-adk-addonkit #320: Windows 10 IOT Problem - BYTx64FM.xml not found

Takaao1, 2019-05-31:
Hello, i have a problem when I try to build an image for a computer who has a x64 processor INTEL.

I'm trying to build an image for a x64 processor, but i don't know why i have this error, by the way i have this file in the folder.
Could you please help me ? I have past a lot of hours trying to find why I have this error.
( I have installed ADK + WDK latest versions, adk-addonkit too)
[image]

parameshbabu, 2019-06-01:
Hi,

Can you follow the instructions for Intel BSP here
https://docs.microsoft.com/en-us/windows-hardware/manufacture/iot/bsphardware#intel-bsps

Regards,
Paramesh

[image]<https://nam06.safelinks.protection.outlook.com/?url=https%3A%2F%2Fuser-images.githubusercontent.com%2F43552973%2F58708745-dc9cde80-83b8-11e9-8872-7e27a523e10c.png&data=02%7C01%7Cpabab%40microsoft.com%7C5b2e42168fa047d0ed8f08d6e5cc4cab%7C72f988bf86f141af91ab2d7cd011db47%7C1%7C0%7C636949062950050610&sdata=kvH3W18vTKbi8v05ZkDNtpo93cB5gO7Hffgg5W673MA%3D&reserved=0>

Takaao1, 2019-06-03:
Geat, it works! 
Thank you very much

closed 2019-06-03 by Takaao1