ms-iot/iot-adk-addonkit #321: Windows 10 IoT start but then black screen

Takaao1, 2019-06-04:
I have installed the application from the factory where i work, on the image .ffu

Then, i have installed the windows PE on the hard drive, and added the .ffu image on the root of it.

At least, i have put the hard drive on the computer, who have launched a cmd console, i have done this command for launch the .ffu file :

dism.exe /apply-image /ImageFile:Flash.ffu /ApplyDrive:\\.\PhysicalDrive0 /skipplatformcheck
Then, i have turn off the computer, and put out the hard drive

I have launched the computer, a Windows 10 IoT page came like this :
[image]

And after, i have a black screen.

I'm turning off / on everytime the computer, but i have still the black screen.

Why ? What Am I supposed to do ???

I'm so confused, i can do nothing and i'm at 99 % of my project now ...

I need your help as quick as possible, because the app from the factory has to work with this Windows 10 IoT Core!

parameshbabu, 2019-06-04:
Hi
App install instructions are here.
https://docs.microsoft.com/en-us/windows/iot-core/develop-your-app/appinstaller

Is your app set as default?

Regards,
Paramesh

[image]<https://nam06.safelinks.protection.outlook.com/?url=https%3A%2F%2Fuser-images.githubusercontent.com%2F43552973%2F58874759-e761b700-86c9-11e9-8f53-671c76927ba6.png&data=02%7C01%7Cpabab%40microsoft.com%7C2f7d55e7da0f4f89bf2508d6e8dd2aaf%7C72f988bf86f141af91ab2d7cd011db47%7C1%7C0%7C636952433931511228&sdata=vJmdGzxlntbd49yH1h69OGS0qE6%2Bi3J479KzgfHE3VY%3D&reserved=0>

Takaao1, 2019-06-04:
I guess yes, i have followed all the instructions of it , so it's why i'm confused because the installation on the computer is not working (still have this black screen).
What am I supposed to do ?

Takaao1, 2019-06-04:
Just , the package application was on .appxbundle, is it a problem ? Is it the cause of the problem ? I was thinking about it

parameshbabu, 2019-06-05:
Appxbundle is not an issue ( except that it bloats the package size for unnecessary binaries for other arch).
Have you tried installing the appx independently and ensure that it works on the device?
Are you building a retail image or a test image?

Takaao1, 2019-06-05:
Yes i tried to install it independatly, but it doesn t work, I have the same problem ( black screen )

Obtenez Outlook pour iOS<https://aka.ms/o0ukef>

parameshbabu, 2019-06-05:
I think the application needs to be debugged on why it’s not launched. This is not an imaging issue. Recommend to manually deploy the app and debug.

Regards,
Paramesh

Takaao1, 2019-06-05:
By the Way it s a test image and not a retail image, and how am i supposed to do it ? Because the application was made by a developer from the factory and she works with other devices, so are you sure ?

Obtenez Outlook pour iOS<https://aka.ms/o0ukef>

Takaao1, 2019-06-05:
And what do you mean by " Recommend to manually deploy the app and debug " ?? 
Now i'm lost.
I don't know exactly what steps i'm supposed to do

parameshbabu, 2019-06-05:
https://docs.microsoft.com/en-us/windows/iot-core/develop-your-app/remotedebugging should help you with your debugging process.

Takaao1, 2019-06-05:
How do you want from me to do that, I can't because i Have a black screen, and the only way for me to have my IoT Core good is to don't put the app on it , so it will be usless to do that without app
So for me, it won't work, right?

parameshbabu, 2019-06-05:
Build the device without your app and boot it and then manually install your app ( see https://docs.microsoft.com/en-us/windows/iot-core/develop-your-app/appinstaller ) and launch to see why your app is not running. You are debug at this time as well. Once you have got your app running on the device then you can create the image including that app.

Takaao1, 2019-06-05:
I built the device without the app, but now when i'm at the home page of Windows 10 IoT Core, i need to access to Wi-Fi (But it found nothing, why ?) And when i try to access to the Windows Device Portal, it doesn't work too
With the username : administrator
and the password : p@ssw0rd

How am I supposed to do to works ?

IsmailHassani, 2019-06-23:
Check this [https://social.msdn.microsoft.com/Forums/en-US/5e69a7eb-b8f6-4ac8-a150-025e7d045842/uwp-application-is-not-installed?forum=WindowsIoT](https://social.msdn.microsoft.com/Forums/en-US/5e69a7eb-b8f6-4ac8-a150-025e7d045842/uwp-application-is-not-installed?forum=WindowsIoT) where i had the same issue. Maybe it helps