ms-iot/iot-adk-addonkit #322: Retail Image Installation : Black Screen

Perosperos, 2019-06-04:
Hello,

i am currently try to create a retail image, but all results in an short boot up with Windows logo and then a black screen appears.

I try to write down all the steps i did:
1.
We bought a kernel mode signing certificate.
Certificate:
Issuer: thawte SHA256 Code Signing CA - G2
Subject: Company
Thumbprint: 030375f518e4f6801419f93bc0ef559fa3d67019

The cross certificate i downloaded from [Link](https://docs.microsoft.com/en-us/windows-hardware/drivers/install/cross-certificates-for-kernel-mode-code-signing)
Cross Certificate:
Issuer: Microsoft Code Verification Root
Subject: thawte Primary Root CA - G3
Thumbprint: ba57ca5e78dd2d1d7476aebee9953e396fd05546

Both certificate were exported to certificate manager (MMC)
2.
I edit IoTWorkspace.xml found on ./iot-adk-addonkit-master\Workspace\IoTWorkspace.xml with:
<RetailSignToolParam>/s my /sha1 "030375f518e4f6801419f93bc0ef559fa3d67019" /fd SHA256</RetailSignToolParam>

Cross Certificate copied in: .\iot-adk-addonkit-master\Workspace\Certs\thawte Primary Root CA.cer

I opened .\iot-adk-addonkit-master\IoTCorePShell.cmd and typed in:

Set-IoTRetailSign On
New-IoTCabPackage All

Redo-IoTCabSignature .\iot-adk-addonkit-master\Workspace\Build\arm\pkgs C:\Users\Company\Desktop\Test\cabs 
xcopy C:\Users\Company\Desktop\Test\cabs\*.cab C:\Users\Company\Desktop\Test\iot-adk-addonkit-master\Workspace\Build\arm\pkgs\*.cab

sign.cmd C:\Users\Company\Desktop\Test\RPi_BSP\Rpi2\Packages\RPi.Drivers\*.dll
sign.cmd C:\Users\Company\Desktop\Test\RPi_BSP\Rpi2\Packages\RPi.Drivers\*.sys

New-IoTWorkspace C:\Users\Company\Desktop\Test Company arm                                                                                
Import-IoTOEMPackage Recovery.*
Import-IoTBSP RPi2 C:\Users\Company\Desktop\Test\RPi_BSP
New-IoTCabPackage All
Add-IoTProduct ProductA RPi2
New-IoTFFUImage ProductA Retail

All steps were sucessful.
After that i installed the image on a MicroSD Card but only the windows logo appears.

Thanks for the help

IsmailHassani, 2019-06-23:
Check this [https://social.msdn.microsoft.com/Forums/en-US/5e69a7eb-b8f6-4ac8-a150-025e7d045842/uwp-application-is-not-installed?forum=WindowsIoT](https://social.msdn.microsoft.com/Forums/en-US/5e69a7eb-b8f6-4ac8-a150-025e7d045842/uwp-application-is-not-installed?forum=WindowsIoT) where i had the same issue. Maybe it helps

matelich, 2019-07-16:
I'm having a similar problem (though on Dragonboard). Did you find a solution?

matelich, 2019-07-17:
https://github.com/ms-iot/iot-adk-addonkit/issues/306 actually had the solution for me, though it took a while for me to realize it. There was a different directory of cab files for my BSP that I hadn't noticed and thus hadn't signed.