ms-iot/iot-adk-addonkit #323: ERROR: PrepareVirtualHardDisk: AttachVirtualDisk failed with error 0x80070522

conekeo, 2019-06-13:
I followed this tutorial to create a basic image of Windows IOT Core:
https://docs.microsoft.com/en-us/windows-hardware/manufacture/iot/create-a-basic-image

I run IoTCorePShell.cmd as an administrator.

All the previous steps was successful, but I encountered an error while running this step: 
```
New-IoTFFUImage ProductA Test
(or) buildimage ProductA Test
```

Here the errors:

```
PS C:\MyWorkspace>New-IoTFFUImage ProductA Test
ADK_VERSION : 10.0.17763.1
IOTCORE_VER : 10.0.17763.253
BSP_VERSION : 10.0.0.0
ADDONKITVER : 6.0.190307.1402
HostOS Info : Microsoft Windows 10 Enterprise - 10.0.17763 - en-US fr-CA
Validating product feature ids
Reading feature ids in C:\MyWorkspace\Source-arm\BSP\Rpi2\Packages\RPi2FM.xml
Reading feature ids in C:\MyWorkspace\Common\Packages\OEMCommonFM.xml
Reading feature ids in C:\MyWorkspace\Source-arm\Packages\OEMFM.xml
Checking Microsoft features in OEMInput file..
Warning: IOT_APPLICATIONS is not defined
Warning: IOT_GENERIC_POP is not defined
Checking OEM features in OEMInput file..
Building product specific packages
Processing Registry.Version.wm.xml
Processing Custom.Cmd.wm.xml
Processing Provisioning.Auto.wm.xml
Processing Recovery.WinPE.wm.xml
Building FM files..
Exporting OEM FM files..
Processing OEMFMList..
Exporting RPi2 BSP FM files
Processing RPi2FMFileList.xml
Creating Image..
See C:\MyWorkspace\Build\arm\ProductA_Test.log for progress
This will take a while...
ThreadId2588 ERROR: PrepareVirtualHardDisk: AttachVirtualDisk failed with error 0x80070522
ThreadId2588 ERROR: CreateVirtualHardDisk: OpenDiskInternal failed with error 0x80070522
ThreadId2588 ERROR: Imaging!CreateImage: Failed to create FFU: CreateVirtualHardDisk failed with error code: 80070522
ThreadId2588 ERROR:     CreateVirtualHardDisk failed with error code: 80070522
Error: Build failed
False
IoTCorePShell arm 10.0.0.0 Test
```

any idea?
Note: The ADDONKITVER is version 1903, I was unable to find 1809.

mMend, 2019-06-17:
I got the same errors, but with code 0x800703e3. There was an IT policy that no new storage media shall be attached. Temporary deactivation solved the problem.

conekeo, 2019-06-17:
@mMend  Thanks for your suggestion;  I updated my Group Policy and  it's working now!

closed 2019-06-17 by conekeo