ms-iot/iot-adk-addonkit #324: Suggested Improvement: IoT-AddFilePackage

kiwikissling, 2019-07-10:
The IoT build command `IoT-AddFilePackage` found in `https://github.com/ms-iot/iot-adk-addonkit/blob/master/Tools/IoTCoreImaging/IoTAddCommands.ps1` does not expose selection of the partition that the file(s) is/are to be deployed to. The `MainOS` partition is assumed, despite the underlying New-IoTWMWrite class supporting explicit targeting of the desired partition.

It would be good to add an optional, validated partition parameter to `IoT-AddFilePackage`, and have this give effect to selecting the desired partition for the file(s)

parameshbabu, 2019-07-10:
Good suggestion. Will look into this. Thanks.

kiwikissling, 2019-07-11:
Great. While you're in that part of the codebase there appears to be a problem with the  [String[][]] $myfiles array passed to this function (and to `Add-IoTRegistryPackage`) ... that, or it could be my lack of understanding about powershell.

I have found that this array must have 2 or more files defined. A single row (ie 3 strings) and the function bombs. Does this need to be fixed, or is there some way to terminate a 2-dimension array with just one row?

parameshbabu, 2019-10-11:
@kiwikissling please feel free to make the change and submit a PR.  Might be a while when I get to this. Thanks in advance.

parameshbabu, 2020-03-23:
> Great. While you're in that part of the codebase there appears to be a problem with the [String[][]] $myfiles array passed to this function (and to `Add-IoTRegistryPackage`) ... that, or it could be [...]

If you pass the array properly it works. For single item, it should still be an array. For example
```powershell
$regkey = @(("`$(hklm.software)\`$(OEMNAME)\Test","StringValue", "REG_SZ", "Test string"))
Add-IoTRegistryPackage Reg.Settings $regkey
```

parameshbabu, 2020-03-23:
> The IoT build command `IoT-AddFilePackage` found in `https://github.com/ms-iot/iot-adk-addonkit/blob/master/Tools/IoTCoreImaging/IoTAddCommands.ps1` does not expose selection of the partition that the file(s) is/are to be deployed to. The `MainOS` partition is assumed, despite the underlying `New-IoTWMWriter` class providing an optional partition parameter.
> 
> It would be good to add an optional, validated partition parameter to `IoT-AddFilePackage`, and have this give effect to selecting the desired partition for the file(s)

In regards to the original ask, the recommendation is always to store all your OS payloads in the MainOS and not target anything to other partitions. Is there an need to target other partitions, let me know the details. For now, closing this.

closed 2020-03-23 by parameshbabu