ms-iot/iot-adk-addonkit #325: OEMInput Features not defined

TheJimmyJ57, 2019-09-24:
When creating my IoT Image, I added the feature IOT_DMAP_DRIVER to the OEMInput.xml. When performing the New-IotFFUImage command, It says that this feature and others are not defined. How can I get them to be defined so I can use them?

```
Warning: IOT_APPLICATIONS is not defined
Warning: IOT_CORTANA is not defined
Warning: IOT_DISABLE_UMCI is not defined
Warning: IOT_GENERIC_POP is not defined
Warning: IOT_NETCMD is not defined
Checking OEM features in OEMInput file..
Warning: IOT_DMAP_DRIVER is not defined
Building product specific packages
Processing Registry.Version.wm.xml
Processing Custom.Cmd.wm.xml
Processing Provisioning.Auto.wm.xml
Processing Recovery.WinPE.wm.xml
Building FM files..
Exporting OEM FM files..
Processing OEMFMList..
Exporting BSWx64 BSP FM files
Processing BSWx64FMFileList.xml
Creating Image..
```

parameshbabu, 2019-09-24:
These are deprecated features. Please see https://docs.microsoft.com/en-us/windows-hardware/manufacture/iot/iot-core-feature-list

closed 2019-09-24 by parameshbabu

TheJimmyJ57, 2019-09-24:
The features list per version says that IOT_DMAP_DRIVER is still included in the 1809 release https://docs.microsoft.com/en-us/windows-hardware/manufacture/iot/iot-core-feature-list#features-per-release

parameshbabu, 2019-09-24:
Yes IOT_DMAP_DRIVER is already defined. Which version are you checking this against?