ms-iot/iot-adk-addonkit #330: Unknown argument for pkggen when executing importbsp

petercoin, 2019-12-09:
Hi,
I tried to create an IoT Core image for Apollo Lake platform and ran into a problem when executing importbsp.
It shew **"Unknown argument "useLegacyName"  "foroempkg"** were provided." but the Copy-IoTBSP function was still completed.
I have also noticed that pkggen seems not support above two parameters from [MSFT's website](https://docs.microsoft.com/en-us/windows-hardware/manufacture/iot/create-packages).
Is this the root cause of the error I met?

parameshbabu, 2019-12-09:
Hi, please use ADK 1809 version for image creation. (https://go.microsoft.com/fwlink/?linkid=2026036) I will update the links to point to 1809 version. The latest versions may not work.

petercoin, 2019-12-10:
Hi,
Just tried but still got error.
Sorry that I use Chinese version Windows and the error message shows some Chinese characters.  

PS D:\IoTCoreImage\MB1>importbsp APLx64 'C:\Program Files (x86)\Intel IoT\Source-x64\BSP\'
Processing C:\Program Files (x86)\Intel IoT\Source-x64\BSP\\APLx64
指定的引數超出有效值的範圍。**(This may be argument out of range)**
參數名稱: Unknown argument "useLegacyName"  "foroempkg" were provided.
Error: D:\IoTCoreImage\MB1\Source-x64\BSP\APLx64\Packages\APL64.CMD\APL64.CMD.pkg.xml pkggen failed
不可在值為 Null 的運算式上呼叫方法。**(Doesn't know how to translate back to English version. It means that I cannot call a method of the object because its value is null.)**
位於 C:\iot-adk-addonkit\Tools\IoTCoreImaging\IoTWorkspace.ps1:856 字元:21
+                 if (-not $pkgnames.contains($definedpkg)) {
+                     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    + CategoryInfo          : InvalidOperation: (:) []，RuntimeException
    + FullyQualifiedErrorId : InvokeMethodOnNull

parameshbabu, 2019-12-18:
Can you provide the details of the OS version you are using? Ideally the first set of details when you launch the prompt first.

petercoin, 2019-12-19:
Hi,
My system information after executing IoTCorePShell.cmd is as below:

Loading IoTCoreImaging module..
Warning: arm IoT Core kit not found.
Warning: x86 IoT Core kit not found.
x64 IoT Core kit found.
Warning: arm64 IoT Core kit not found.
Test certs installed
Opening workspace : C:\iot-adk-addonkit\Workspace\IoTWorkspace.xml
Error: arm IoT Core kit not found.
ADK_VERSION : 10.0.17763.1
IOTCORE_VER : 10.0.17763.253
BSP_VERSION : 10.0.0.0
ADDONKITVER : 6.0.191010.1210
HostOS Info : Microsoft Windows 10 Pro - 10.0.18363 - zh-TW en-US
IOTWKSPACE  : C:\iot-adk-addonkit\Workspace
OEM_NAME    : Contoso
BSP_ARCH    : arm
BSPPKG_DIR  : C:\iot-adk-addonkit\Workspace\Build\arm\pkgs
MSPKG_DIR   : C:\Program Files (x86)\Windows Kits\10\MSPackages\Retail\arm\fre
IoTCorePShell arm 10.0.0.0 Test

The iot-adk-addonkit is cloned from Github and is updated to the latest version.

blknit, 2020-02-28:
I have the same issue. Any update?

parameshbabu, 2020-03-06:
The workspace architecture is arm while you are looking to build amd64. Please do "setenv x64" to setup the proper architecture and recreate the workspace. That should solve the build issue.

closed 2020-03-06 by parameshbabu