ms-iot/iot-adk-addonkit #332: Adding custom app

sebillen, 2019-12-10:
Hi,
Yesterday, I upgrade tools with last version and I found issue with Add-IoTAppxPackage
I create my Appx package with VisualStudio 2019 and obtain myApplication_1.0.0.0_ARM_Debug.msix and not myApplication_1.0.0.0_ARM_Debug.appx
with commits on Nov 14, 2018 Add-IoTAppxPackage accept *.msix and create component.
With last commits Add-IoTAppxPackage don't accept and generate error.

What do I make wrong?

parameshbabu, 2019-12-18:
can you give more details on the error you see? Thanks.

sebillen, 2019-12-19:
Hi,
When following 'Lab 1b: Add an app to your image' https://docs.microsoft.com/en-us/windows-hardware/manufacture/iot/deploy-your-app-with-a-standard-board

I create my own UWP application (Visual Studio 2019), create App Packages, obtain folder with
MyUWPApp_1.0.1.0_ARM_Debug.msix + other files

**_On iot-adk-addonkit commit Nov 14, 2018_**
**command:** Add-IoTAppxPackage "C:\Users\xxx\source\Repos\AzureDemo\IoTDemo\AppPackages\MyUWPApp_1.0.1.0_ARM_Debug_Test\MyUWPApp_1.0.1.0_ARM_Debug.msix" fga Appx.MyUWPApp
**Result**:
AppxName          : MyUWPApp
AppxVersion       : 1.0.1.0
PackageFamilyName : MyUWPApp_50h5jfts8t0pr
Dependencies      : Microsoft.NET.CoreFramework.Debug.2.2.appx Microsoft.NET.CoreRuntime.2.2.appx Microsoft.VCLibs.x64.Debug.14.00.appx
Creating C:\Users\xxx\Documents\Embedded\images\Win10IoTCore\Ltsc2019\Msc\Source-x64\Packages\Appx.MyUWPApp\customizations.xml...
Processing MyUWPApp.cer with thumbprint 696241BF33C70E79F9F8E3A5AF55CAD330A8645B
Method invocation failed because [System.String] does not contain a method named 'AppendChild'.
At C:\iot-adk-addonkit_1809_10.0.17763_20181120\Tools\IoTCoreImaging\Classes\IoTProvisioningXML.ps1:321 char:13
    +$node.AppendChild($newnode)
    +~~~~~~~~~~~~~~~~~~~~~~~~~~~
    + CategoryInfo          : InvalidOperation: (:) [], RuntimeException
    + FullyQualifiedErrorId : MethodNotFound

New Appx created at C:\Users\xxx\Documents\Embedded\images\Win10IoTCore\Ltsc2019\Msc\Source-x64\Packages\Appx.MyUWPApp
Feature ID : APPX_MYUWPAPP
IoTCorePShell amd64 10.0.0.0 Test

Got one error but package created and can continue Lab.

_**On last Commit**_
**command:** Add-IoTAppxPackage "C:\Users\xxx\source\Repos\AzureDemo\IoTDemo\AppPackages\MyUWPApp_1.0.1.0_ARM_Debug_Test\MyUWPApp_1.0.1.0_ARM_Debug.msix" fga Appx.MyUWPApp
**Result**:
Error: C:\Users\xxx\source\Repos\AzureDemo\IoTDemo\AppPackages\MyUWPApp_1.0.1.0_ARM_Debug_Test\MyUWPApp_1.0.1.0_ARM_Debug.msix is not an appx/appxbundle

cannot continue Lab

Regards,

sebillen, 2020-02-12:
Hi, any news?

parameshbabu, 2020-03-06:
What version of the ADK tools are you using? Can you provide the info dumped when you launch the iot core shell?

sebillen, 2020-03-06:
Hi,
I use iot-adk-addonkit commit on Oct 16, 2019
iot core shell lunch time:

Loading IoTCoreImaging module..
Opening workspace : C:\iot-adk-addonkit_1809_10.0.17763_20191016\Workspace\IoTWorkspace.xml
ADK_VERSION : 10.0.18362.1
IOTCORE_VER : 10.0.17763.253
Corekit found OK
Test certs installed
IOTWKSPACE  : C:\iot-adk-addonkit_1809_10.0.17763_20191016\Workspace
OEM_NAME    : Contoso
BSP_ARCH    : arm
BSP_VERSION : 10.0.0.0
BSPPKG_DIR  : C:\iot-adk-addonkit_1809_10.0.17763_20191016\Workspace\Build\arm\pkgs
MSPKG_DIR   : C:\Program Files (x86)\Windows Kits\10\MSPackages\Retail\arm\fre
IoTCorePShell arm 10.0.0.0 Test
PS C:\iot-adk-addonkit_1809_10.0.17763_20191016\Workspace>

Regards,

parameshbabu, 2020-03-06:
Please use the ADK version 1809 (ADK_VERSION : 10.0.17763.1). This is the version supported with IoT Core.

closed 2020-03-06 by parameshbabu