ms-iot/iot-adk-addonkit #333: OEM.Sample Package

mawett, 2019-12-16:
While cleaning up the packages in our workspace I came across "OEM.Sample.wm.xml". For me it just looked like sample on how to add registry key to an image, so I removed it.
However, now my builds fail when creating a new workspace, because the OEM.Sample package can't be found.
It seems that workspace creation sees this as a required packaged, but I have no idea why:
```
    #Import required packages
    Import-IoTOEMPackage Registry.Version
    Import-IoTOEMPackage Custom.Cmd
    Import-IoTOEMPackage Provisioning.Auto
    **Import-IoTOEMPackage OEM.Sample**
```

The first three make sense to me, but I don't see why OEM.Sample would be there. I will remove it for now, but it would be good to have a confirmation about what the intention was of adding it there and that I am not missing anything.

parameshbabu, 2019-12-18:
This was added to make sure that there is something in the oemfm.xml file. You may just ignore this or remove this package and also its reference in the oemfm.xml file. No harm doing this.

closed 2019-12-18 by parameshbabu

mawett, 2019-12-19:
Great, thanks for clarifying.