ms-iot/iot-adk-addonkit #336: Fatal Error 0x80188304 while building the image stucking at 97%

HnourN, 2020-02-20:
Hello,

I am getting always these errors while building thr image for an Iot board x86 E3900 series, the problem is occuring when reaching 97%. Can someone tell me pls you are familiar with this kind of errors ?

```
ThreadId9188 ERROR: [00:07:25] Operation failed finalizing changes: 80188304 [Error,Facility=(0018),Code=33540 (0x8304)]
ThreadId9188 ERROR: [00:33:10] onecore\base\cbs\mobile\iucore\iupartition.cpp, IUPartition::ProcessIndividualPackages, line 987, Error  , Failed to finalize client, error is 0x80188304
ThreadId9188 ERROR: [00:33:10] onecore\base\cbs\mobile\iucore\iupartition.cpp, IUPartition::ProcessAllPackages, line 1066, Error  , Failed to process individual packages while Commit, error is 0x80188304
ThreadId9188 ERROR: [00:33:10] onecore\base\cbs\mobile\updatedll\lib\executeupdate.cpp, UpdateMain::Commit, line 1031, Error  , Failed to commit partition MainOS, error is 0x80188304
ThreadId9188 ERROR: [00:33:17] onecore\base\cbs\mobile\updatedll\lib\executeupdate.cpp, UpdateMain::ExecuteUpdate, line 864, Error  , hr = 0x80188304
ThreadId9188 ERROR: Imaging!UpdateImage: Failed call to UpdateDLL::ExecuteUpdate with error code: -2145877244 (0x80188304)
Error: Build failed
```

parameshbabu, 2020-03-06:
Can you provide the system setup info completely? The first set of info when you launch the iot core shell.

HnourN, 2020-03-06:
Yes of course, note that the arm IoT kit was installed during the image creation.
Intel Atom Processor BSP's x86 version 5.3.4.0808

```
Loading IoTCoreImaging module..
Warning: arm IoT Core kit not found.
x86 IoT Core kit found.
x64 IoT Core kit found.
Warning: arm64 IoT Core kit not found.
Test certs installed
Opening workspace : C:\Users\********\Downloads\IOTCore\iot-adk-addonkit-master\Workspace\IoTWorkspace.xml
Error: arm IoT Core kit not found.
ADK_VERSION : 10.0.17134.1
IOTCORE_VER : 10.0.15063.0
BSP_VERSION : 10.0.0.0
ADDONKITVER : 6.0.191010.1210
HostOS Info : Microsoft Windows 10 Pro - 10.0.17763 - en-US
IOTWKSPACE  : C:\Users\********\Downloads\IOTCore\iot-adk-addonkit-master\Workspace
OEM_NAME    : Contoso
BSP_ARCH    : arm
BSPPKG_DIR  : C:\Users\********\Downloads\IOTCore\iot-adk-addonkit-master\Workspace\Build\arm\pkgs
MSPKG_DIR   : C:\Program Files (x86)\Windows Kits\10\MSPackages\Retail\arm\fre
IoTCorePShell arm 10.0.0.0 Test

```

parameshbabu, 2020-03-06:
The IoTCore kit version you are using is old (IOTCORE_VER : 10.0.15063.0). Please uninstall that version and update to the latest available version.  Also the ADK version needs to be updated as well (ADK_VERSION : 10.0.17763.1).

closed 2020-03-06 by parameshbabu