ms-iot/iot-adk-addonkit #346: Deprecated Timestamp Server

MattBranch, 2021-01-13:
When building a retail image of IoT Core, the `buildimage` command fails when attempting to contact the now deprecated "http://timestamp.verisign.com/scripts/timstamp.dll" Verisign URL (per "FIPPackage_oem.log", "FIPPackage_ocp.log", and "FIPPackage_QCDB410C.log").  I have edited the `TIMESERVER` variable in "sign.cmd" to point to "http://timestamp.digicert.com" which has allowed the `buildpkg` steps to complete successfully, but the final `buildimage` step fails as it is still pointing to the Verisign server.  How can I point any code signing steps to the new Digicert server?  This is a critical issue as we need a new retail image for production ASAP.  Any help you could provide would be most appreciated.

parameshbabu, 2021-01-13:
@MattBranch taking a look at this now and will revert back.

parameshbabu, 2021-01-13:
As a quick fix, please also change the same at this file location
C:\Program Files (x86)\Windows Kits\10\Tools\bin\i386\sign.cmd
line 617 
if "%SIGN_WITH_TIMESTAMP%"=="1" set TIMESERVER=/t http://timestamp.verisign.com/scripts/timestamp.dll

parameshbabu, 2021-01-13:
@MattBranch  the timeserver editing in the above mentioned place should fix the issue.

MattBranch, 2021-01-13:
@parameshbabu , thank you for helping with this so quickly.  Much appreciated!  Manually editing the "sign.cmd" file at the location you specified did the trick, on top of editing the "sign.cmd" that was part of my build workspace.  My image has now built successfully.

I have not tried the permanent fix you put in place, but one thing I noticed was that you used "http://timestamp.digicert.com/scripts/timestamp.dll" as the new timestamp URL.  My testing was with "http://timestamp.digicert.com", as Digicert's documentation shows (see "https://www.digicert.com/kb/code-signing/ev-authenticode-certificates.htm").  If the extended URL worked for you, I believe we can close this issue now.  Thank you once again!

parameshbabu, 2021-01-13:
@MattBranch Thanks for the details. The fix i put in is same as yours and the sign.cmd on the tools dir needs to be manually patched. I'm following up internally for a better solution.

Thanks for the link, i have used this deeper path and it does work properly. Closing this issue now.

closed 2021-01-13 by parameshbabu