ms-iot/iot-adk-addonkit #348: Environment configuration problem "Device Guard"

527395632, 2021-01-26:
Open "IoTCorePShell", as shown in the following figure:
[image]

PowerShell Print Log: 
------------------------------------------------------------
Loading IoTCoreImaging module..
ADK version 10.0.17763.1 found
arm IoT Core kit version 10.0.17763.253 found
Warning: x86 IoT Core kit not found
Warning: x64 IoT Core kit not found
Warning: arm64 IoT Core kit not found
Test certs installed
Opening workspace : D:\iot-adk-addonkit\Workspace\IoTWorkspace.xml
arm IoT Core kit version 10.0.17763.253 found
Corekit install path found OK
ADK_VERSION : 10.0.17763.1
IOTCORE_VER : 10.0.17763.253
BSP_VERSION : 10.0.0.0
ADDONKITVER : 6.0.200326.1705
HostOS Info : Microsoft Windows 10 企业版 LTSC - 10.0.17763 - zh-CN
Error: Host OS Enterprise Edition required for using Security functions such as Device Guard
IOTWKSPACE  : D:\iot-adk-addonkit\Workspace
OEM_NAME    : Contoso
BSP_ARCH    : arm
BSPPKG_DIR  : D:\iot-adk-addonkit\Workspace\Build\arm\pkgs
MSPKG_DIR   : C:\Program Files (x86)\Windows Kits\10\MSPackages\Retail\arm\fre
IoTCorePShell arm 10.0.0.0 Test
PS D:\iot-adk-addonkit\Workspace>
------------------------------------------------------------
be careful: **Error: Host OS Enterprise Edition required for using Security functions such as Device Guard**

The general meaning of this problem is to enable the virtualization security function of hyper-v. I searched a lot of materials and referred to the methods of the following websites：

A: https://www.tenforums.com/tutorials/68913-enable-disable-device-guard-windows-10-a.html?__cf_chl_jschl_tk__=d1fcf90156c4b1672ce2e6c72b0d2658628f8e99-1611626151-0-AQYhrR5jAKpae59wP4nNcrU2hiows8OegWaPt-1dQNQZRIeJufBaTNNrZrWRV4Pu_6WO9__Y6svwp270Q2cqLBsahjRZWjQ1OKVMl1mrG-yXGIa2MTKR2WG5HwUHsX_vuJnFO5s7do-zp-19Nt34waWs-Up4LohtfLsvN71-4AFBbR13Mp1S88x8oviaFhltxGYhzEPyF4W8hg-SOTTB7eILhlzUAR6EahktCJ6WnDoUbnX_8g-01pciTs9xzFT97i8PifGE5OQo0EIOJQRo4xzHh_1mknPnUBOZjmG3wS2jUmC69TuzIOjaB_s8Zt-alO-fYiV1-FFvx9NqFOnihRrsiQ_W1LlCj1dz2L-nkyDNpMkDqsjYOWplCt_NK0yBla0Y_Is3eunIB34f87HGvEc

B: https://techcommunity.microsoft.com/t5/iis-support-blog/windows-10-device-guard-and-credential-guard-demystified/ba-p/376419

According to the above method, I did the following operations：
1、The CPU virtualization function of "VMware Workstation" is turned on
[image]
[image]

2、Hyper-V hypervisor installed
[image]

3、Enable turn on virtualization based security
[image]
[image]

The problem is still unresolved. I hope I can get help.

parameshbabu, 2021-01-26:
The issue reported is for using DeviceGuard powershell scripts on the PC and not really about the device guard feature in the host PC. 
In this specific case, since you are already using LTSC, you can ignore this warning. The script should be throwing this error if its not Enterprise or LTSC.

To summarize,you can ignore this warning as you are already running a LTSC edition ( which is enterprise ).

527395632, 2021-01-27:
Oh, that's right. Thank you very much. In addition, when will Raspberry Pi 4 be supported? Raspberry Pi CM4 is very suitable for making products. I'm looking forward to his joining