ms-iot/iot-adk-addonkit #361: Certificate chains

matelich, 2022-09-07:
The cert we had been using for creating retail builds expired and we use a homegrown tool which launches ApplyUpdate -stage [cab], ApplyUpdate -commit to perform updates of our software. I'm trying to get a new update out the door and my current cert I'm using is failing with "A certificate chain could not be built to a trusted root authority."

I'm not sure how to determine if I'm fully out of luck because there would be no acceptable root authorities, or if I just need a different cert. Original was Verisign, new is Digicert.

I'd love to be able to keep shipping updates to devices with my FFU built in 2019 (10.0.17763.253).

parameshbabu, 2022-09-08:
Hi, we have recommended to move to SHA2 based signing (see https://github.com/ms-iot/iot-adk-addonkit#17763-v7-branch) as the sha1 certificates are expiring/expired and not planned to be supported further. Can you see the instructions in the link and update your devices?

matelich, 2022-09-08:
I have looked through that a bit, what I don't see is instructions for
updating an existing product line. My devices do not support user
installation of a new os. I guess I was looking for confirmation
(preferably not 😁) that I cannot generate new software to be installed on
my existing fleet.

saapte, 2022-10-11:
@matelich For migration of existing devices from cross-signed > custom signed binaries, follow the same steps as for a new FFU (except the FFU generation). Once the v7 versions of the Secure Boot, Device Guard, and your custom cert signed packages are ready, they can be deployed via Device Update Center, or manually using applyupdate.exe.

A few things to be careful with:
1. On the latest IoT builds, it's imperative that you build and deploy both the new version of secure boot and device guard when migrating to custom signed binaries, and not device guard alone
2. Please sure that the host you use to build these packages is running Enterprise with major version 17763 (1809), and not 21H2 or some other version.