ms-iot/iot-adk-addonkit #363: v7 issues

matelich, 2022-09-21:
Since I don't want to have to make a new FFU in a year after the replacement certs expire, I'm trying to switch to the self-signed system as recommended.

I've followed all the steps and I can successfully launch a test build with SEC_BITLOCKER and SEC_SECUREBOOT_TEST. However, if I activate SEC_DEVICEGUARD_TEST the image freezes. Initially I hadn't provisioned the RPMB, but even after that it still doesn't work.

For the prod build, I have not had any success. It freezes after clearing the initial text in the top right, showing my boot image and two underscores.

Would you recommend I open a support ticket? Or do you have any suggestions?

Also, I've previously used the SAC builds, but both links on https://www.microsoft.com/en-us/software-download/windows10iotcore seem to give the same file, so I'm not sure if there's something special that would change my build process such that LTSC would work properly because I've configured it. There isn't really any instructions that tell me how to change my FFU for LTSC mode. I was avoiding going through all the IOT Core Services work until it was necessary, but perhaps that time is now?

sukusuma, 2022-09-23:
Please reach out here to open a ticket with Microsoft.
https://developer.microsoft.com/en-us/windows/support/?tabs=Contact-us

matelich, 2022-09-23:
I have gone that route, thanks. Will update this when a solution is discovered.